ArduPilot/ardupilot #10981: wrong avoiding direction - no option to change

Fungraphics, 2019-04-03:
Using Pixhawk clone on skidsteering rover. Latest stable 3.5 version allows using a dual rangefinder avoidance in steering and auto mode, as it was a feature on old discountinued APM. However it does not work the same way anymore. Instead of using left rangefinder to avoid to the right and right rangefinder to avoid to the left, it always avoids to the same side.

### Feature request

Not sure if it is a bug but the problem is, that both of the rangefinders result in turning the rover the same direction. Always avoiding to the right instead of: left one to the right and right one to the left. It would be more correct to avoid the obstacle by turning the oposite direction away from it.

In the parameters list the : RNGFND_TURN_ANGL has only one value for both rangefinders which is fine, but also have a prefix which is a problem, as there is no way to assign minus for one and plus for another. The feature would be to include RNGFND2_TURN_ANGL as a parameter so one could assign another avoiding direction for the other rangefinder. 

But as i mentioned: on old 8 bit APM (2.4.7) it was handled in the code, so user just installed left and right sonar correctly and one of them was turning the wehicle at one and the other to another side.And as i recall the avoiding turn angle parameter had no prefix.

[ x ] Rover

magicrub, 2019-04-03:
There's a whole new avoidance library about to go in soon. There's an experimental branch now that works only on plane, as described in the below video, but there's work underway to generalize it for all vehicles. 
https://www.youtube.com/watch?v=RCwm95v5j68

closed 2019-04-03 by magicrub

Fungraphics, 2019-04-03:
Thanks a lot.

rmackay9, 2019-04-04:
@Fungraphics, we haven't changed the "Dodge" method of avoidance so it should still continue to turn left or right depending upon which range finder senses the obstacle.  In case it is of interest [the code is here](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/sensors.cpp#L153).

I haven't tested Dodge too much so it's possible that it has broken but it's also possible that it's a setup issue.  Our [wiki page on this feature is here](http://ardupilot.org/rover/docs/rover-object-avoidance.html) but you've probably already seen it.

As TomP suggests, personally I think the Dodge method is quite simple and is unlikely to work in most situations so I'm much more keen to get the advanced solution working.