ArduPilot/ardupilot #10989: Support for RC buttons events

lucasdemarchi, 2019-04-04:
**Platform**
[ X ] All

I'd like to discuss the best way to support buttons normally found on "IP-based Remote Controllers". I recently started a project to run on RCs, particularly on the Sky Controller 2 that comes with the Parrot Disco and implemented something that I think may also work on other RCs: https://github.com/lucasdemarchi/dema-rc/pull/1

On Ardupilot right now we have support for "position switches" so we divide the range (normally from ~1000 to ~2000) and each position maps to a value (range) in this range. 

Contrary to position switches, buttons may have a very short toggle time; it would also be nice to be able to allow buttons to be pressed simultaneously. And to assign arbitrary actions to them.

Questions: does the bitmask concept make sense? Any flaw in that I'm not seeing?  Intention is that after implementing the bitmask on the ardupilot side, we would create actions such as "start streaming", "start recording" and each vehicle may provide a way to do that. Alternative is the virtual relay being added in #10829 - or I could just figure out a way to integrate both.

Another alternative to mapping the buttons to RC channels is to make dema-rc to also talk the mavlink protocol. Then the requests for buttons would come on this "side channel". It seems simpler, but it basically mandates having an additional communication channel (so we can't extrapolate this solution to other types of RCs).

In other products I've seen using buttons and Ardupilot:
  - skyviper: the implementation uses 3 channels and ties to the "Toy Mode", that may not be the best thing to bring to other vehicles
  - Solo: map buttons to individual channels AFAIR.

Thoughts?

peterbarker, 2019-04-04:
On Wed, 3 Apr 2019, Lucas De Marchi wrote:

> Contrary to position switches, buttons may have a very short toggle time;
> it would also be nice to be able to allow buttons to be pressed
> simultaneously. And to assign arbitrary actions to them.

... and to have them both as momentary switches and toggle switch...

lvale, 2019-04-04:
@khancyr also usable with the skycontroller3

@lucasdemarchi you are using "only" the default SC2 events, so no change on the SC2 side? right?

lucasdemarchi, 2019-04-04:
@lvale I don't understand the question. I do have to modify the software running on SC2.  As far as I know it should work as is on SC3. If not, then it should be easy to adapt.

My first try was to use the software that comes with SC2. I decoded the packets and implemented the ArduPilot side. That didn't work well because the same program that controls the buttons/sticks is the one controlling the WiFi network. And it  needs a handshake/keepalive with their flight stack, otherwise it disconnects. So I went to the second approach and decided to do it in a generic way to be applied for other controllers. It's on my to-do list to do a blog post about hacking this controller.

lucasdemarchi, 2019-04-04:
@lvale btw, for disco to work with the software I did, all we need for the sticks is what I sent on #10961

DavidIngraham, 2019-04-13:
@lucasdemarchi How is this different from the AP_JSButton library currently used for Sub?

https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_JSButton

lucasdemarchi, 2019-04-13:
First, I didn't know about this handling on ardusub. Skimming through the code, the library you point is the mapping from buttons to functions, which is similar to what I want. However in your case the button press/releases come mavlink. Here my idea was to provide a way for button events when there's no mavlink channel, only the RC.

tridge, 2019-07-05:
@lucasdemarchi the reason I went with the whole toy_mode thing for skyviper was to minimise impact across the code base, so I could track changes in copter without constant conflicts. It wasn't because it was the best design.
So we really need to decide if we will somehow map buttons to rc channels (eg. by turning a press on a button into a persistent change on a rc channel) or if we will add top level support for buttons.
Mapping to rc channels is the easier path, but not necessarily the better one.
Perhaps we could extend AP_Button?

peterbarker, 2019-07-06:
On Fri, 5 Jul 2019, Andrew Tridgell wrote:

> Perhaps we could extend AP_Button?

Regard should be made for the mavlink messages dealing with remote 
buttons.  I don't know if it will mesh will with my RC_Channel suggestion 
below.

Should also see what the sub guys do.

I'd be leaning towards using an option bit on an RC_Channel to indicate if 
it is toggled with a button.

lucasdemarchi, 2019-07-09:
I think the most straightforward thing to do is to extend the RCInput_UDP protocol to allow communicating the button handling.  This would solve my problem and of any IP-based remote. Those that usually have buttons.

> Should also see what the sub guys do.

See my comment above. The handling in ArduSub is the mapping from event to function. The controller sends the event through mavlink. I'm hesitant on the idea of encapsulating RC in mavlink.

> I'd be leaning towards using an option bit on an RC_Channel to indicate if it is toggled with a button.

You mean a parameter so we know that RC channel is actually a bitmask of buttons? This would allow us to stash 11 buttons, or maybe 8 would be sufficient to account for a few microseconds of jitter  and allow to work on non-IP radios.

IamPete1, 2021-08-15:
I think this is resolved by https://github.com/ArduPilot/ardupilot/pull/16653

You can now activate all the supported RC switch options over MAVLink.

lucasdemarchi, 2021-08-17:
@IamPete1  I'm not sure it's the same thing. You're saying the same aux functions should be configured and then the RC would send a mavlink package (thus over another communication channel) when a button is pressed/released?

In the case for Disco + SkyController, the mavlink stream is redirected from the controller to GCS, so the userspace side doesn't even see the mavlink packets. Yes, we could open another connection, but here I was hoping to just reuse the same channel I'm already sending.

peterbarker, 2021-08-17:
Yes, still pretty sure what we want here is to add a bitmask option parameter into `const AP_Param::GroupInfo RC_Channel::var_info[]` and to make the first Bit `IsToggled`.

Instead of the channel value being set (for the purposes of auxillary function execution) by the absolute value we receive on the channel, each time the channel is asserted (high or low) we move through the possible values for the function assigned to the channel positions.  Or we just ignore the middle position.

IamPete1, 2021-08-17:
I think I might have misunderstood. 

These buttons would be assigned to cannels of a normal RC protocol? So we would still get a number of about 1000 - 2000us? 

If that is the case why we should track the toggling in AP rather than leave it to the transmitter? 

I guess tracking in AP would allow us to support aux functions that are level triggered rather than edge triggered?