ArduPilot/ardupilot #10992: Multi voltage batterie support (i.e. use different battery types but not at the same time))

githubtd01, 2019-04-04:
I would like to (and I think that is intersting for a lot peoples) to use batteries which differ in regard to their voltage discharge curve/behaviour and intitial voltage.

For example: I would like to use 3S and 4S batteries on my copter as well as LiPos and Li Ion batteries.

I think  it would be very easy to implement a function which detects the voltage at startup and changes the parameter (failsafe but also tuning) accordingly. This could at least differenciate between fully charged 3S and 4S batteries. I don't now whether it is possible to differentiate between LIPos and Li-Ions batteries too.

Another idea would be to implement that into Mission Planner by introducing a fast upload function for predefined parameter sets (@meee1 what do you think?).

SergeyBokhantsev, 2019-04-04:
I had an idea that there could be a file-based profiles (on SD card) that user could simply apply either by mavlink comand or by RC stick gesture or per another logic as the battery voltage. These profiles containing the override values for a desired parameters: FS parameters, PIDs, whatever.
https://github.com/ArduPilot/ardupilot/issues/6680

WickedShell, 2019-04-04:
> I think it would be very easy to implement a function which detects the voltage at startup and changes the parameter (failsafe but also tuning) accordingly. This could at least differenciate between [...]

This unfortunately isn't an option, as distinguishing pack type can't be assured simply because we can't make the assumption you plug a full battery in. A charged 5S pack, and a low 6S pack can be indistinguishable based upon voltage. (When you introduce LiIon's, or other less common chemistry's) the situation gets even worse.

A GCS could be used however to swap between values as you suggest. I don't think it's the place for the autopilot to be making these decisions.

@SergeyBokhantsev because of the UI issues I don't see a path where we would bring that feature directly onboard. The closest feature I think that is coming for this, is you could use a one shot script that runs on startup to manipulate the parameter values based on the battery voltage. It's still not something I'd recommend however, as it relies upon the assumption of a full battery being installed into the system.

If there are no objections, I'm probably going to close this as an issue to open against your respective GCS of choice for the fast parameter swap, as that is the only real actionable path forward that I see here.

githubtd01, 2019-04-19:
ok, thank you, I will reopen an issue at the Mission Planer page.

closed 2019-04-19 by githubtd01