ArduPilot/ardupilot #10995: Problem running JSBSIM

Akshath-Singhal, 2019-04-04:
The wiki seems to be incomplete for running SITL ArduPlane with JSBSim. 
The support also seems to be somewhat broken.

Cloned, built and installed jsbsim.
When running SITL with jsbsim using the following command:
`cd ardupilot/ArduPlane; sim_vehicle.py -f jsbsim`, the xterm shows JSBSim to be running while mavproxy reports Link down.
[image]

I am using Ubuntu 16.04 LTS
@magicrub

magicrub, 2019-04-04:
Please link to the specific wiki page with the problem.

Akshath-Singhal, 2019-04-04:
http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#using-jsbsim

Setting up of environment variables is missing which needs to be performed using:
$gedit .bashrc
Adding following lines to the end of bashrc:
export PATH=$PATH:$HOME/jsbsim/build/src
export PATH=$PATH:$HOME/ardupilot/Tools/autotest
export PATH=/usr/lib/ccache:$PATH
And reloading using
$ ~/.bashrc

The above commands solve the JSBSim not found error.
However the error depicted in the image earlier still remains unsolved.

magicrub, 2019-04-05:
Does your simulator work normally if you use the default simulator (don't specify -f jsbsim). Also, try running simvehicle From the plane directory and without the -v.

Akshath-Singhal, 2019-04-05:
Yes it runs well without using -f jsbsim option. I have also tried running it directly from plane directory without -v command but the error still remains same.

peterbarker, 2019-04-05:
On Fri, 5 Apr 2019, Akshath-Singhal wrote:

> Yes it runs well without using -f jsbsim option. I have also tried running
> it directly from plane directory without -v command but the error still
> remains same.

Are you using Ubuntu Dingo64 by any chance?

Akshath-Singhal, 2019-04-06:
> On Fri, 5 Apr 2019, Akshath-Singhal wrote: Yes it runs well without using -f jsbsim option. I have also tried running it directly from plane directory without -v command but the error still remains same.
> Are you using Ubuntu Dingo64 by any chance?

No. I am using UBUNTU 16.04 LTS.

rajat2004, 2019-04-19:
I just tried using JSBSim with an up-to-date master, everything's working perfectly
Regarding exporting the path, I have opened https://github.com/ArduPilot/ardupilot_wiki/pull/1717

rajat2004, 2019-05-20:
The info about adding JSBSim to the PATH has been added in the Wiki - https://github.com/ArduPilot/ardupilot_wiki/pull/1717 and also in the `buid-jsbsim.sh` script - https://github.com/ArduPilot/ardupilot/pull/11160

The problem with the connection was due to JSBSim not shutting down while closing the SITL. This has been fixed in `sim_vehicle.py` - https://github.com/ArduPilot/ardupilot/pull/11342

This issue can be closed now I think

MarceloSemO, 2019-06-14:
I am still facing the problem mentioned here, despite of using the newest version of ardupilot code. The xterm shows JSBSim to be running while mavproxy reports Link down.

closed 2021-07-26 by IamPete1

hdbalyozsht, 2023-01-20:
Same problem.

command: `sim_vehicle.py -v ArduPlane -f jsbsim`
```bash
Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (1.374s)
SIM_VEHICLE: Using defaults from (../../projects/ardupilot/Tools/autotest/default_params/plane-jsbsim.parm)
SIM_VEHICLE: Run ArduPlane
SIM_VEHICLE: "/home/prihex/projects/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduPlane" "/home/prihex/projects/ardupilot/build/sitl/bin/arduplane" "-S" "--model" "jsbsim" "--speedup" "1" "--slave" "0" "--defaults" "../../projects/ardupilot/Tools/autotest/default_params/plane-jsbsim.parm" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501"
RiTW: Starting ArduPlane : /home/prihex/projects/ardupilot/build/sitl/bin/arduplane -S --model jsbsim --speedup 1 --slave 0 --defaults ../../projects/ardupilot/Tools/autotest/default_params/plane-jsbsim.parm -I0
Connect tcp:127.0.0.1:5760 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> link 1 down
```

.bashrc
```bash
export PATH=$PATH:$HOME/jsbsim/build/src
source /home/prihex/projects/ardupilot/Tools/completion/completion.bash
export PATH=$PATH:$ARDUPILOT_PATH/Tools/autotest
export PATH=/opt/gcc-arm-none-eabi-10-2020-q4-major/bin:$PATH
export PATH=/usr/lib/ccache:$PATH
```

jsbsim src directory successfully builded
```bash
~/jsbsim/build/src$ l
CMakeFiles/          GeographicLib/   input_output/  libJSBSim.a  math/    simgear/
cmake_install.cmake  initialization/  JSBSim*        Makefile     models/
```

ParasInternKhushPatil, 2023-04-07:
The Issue still persists 
-------------- begin dumpcore.sh output ----------------
Could not attach to process.  If your uid matches the uid of the target
process, check the setting of /proc/sys/kernel/yama/ptrace_scope, or try
again as the root user.  For more details, see /etc/sysctl.d/10-ptrace.conf
ptrace: Operation not permitted.
/tmp/gdb.21984:2: Error in sourced command file:
You can't do that without a process to debug.
-------------- end dumpcore.sh output ----------------

This is output on xterm window

tz120012, 2024-08-04:
You can try to delete the JSBSim file in the /usr/games/ directory.

tz120012, 2024-08-04:
If FlightGear is installed, you can try to delete the JSBSim file in the /usr/games/ directory.

rananat, 2024-10-17:
> If FlightGear is installed, you can try to delete the JSBSim file in the /usr/games/ directory.

The same issue. Just renamed the JSBSim file. It worked.

rananat, 2024-10-17:
> If FlightGear is installed, you can try to delete the JSBSim file in the /usr/games/ directory.

For the same issue, renaming the JSBSim file /usr/games/ directory resolved it. Thank you @tz120012