ArduPilot/ardupilot #11008: RCx_OPTION=13 causes startup in Simple mode

atshort, 2019-04-05:
### Bug report

Setting RC7_OPTION=13 to enable switching Simple and Super Simple modes from the transmitter causes copter to start up with Simple mode enabled. This is on 3.7. The log shows Simple mode enabled very early (~2 seconds from start of log), before any RCIN values are logged. When RCIN logging begins the log shows a clean 982 for the Simple mode channel - there is no sign of radio glitching, and the transmitter is clearly sending 982 which should be Simple mode = OFF. The switch for C7 correctly switches Simple mode on and off.

This happens with a Holybro Kakute F7 fight controller, an FrSky receiver (R9 Mini and R-XSR tested) plugged into serial 6 (SBUS and S.Port), powered from USB, nothing else connected.

The problem only happens when there is a receiver connected to the flight controller and the transmitter is powered on before the receiver and flight controller.

To reproduce, reset all parameters on the board, then load the following parameters:

SIMPLE,0
SUPER_SIMPLE,0
RC7_OPTION,13
LOG_DISARMED,1
SERIAL6_OPTIONS,7
SERIAL6_PROTOCOL,10

Without the RC7_OPTION,13 line the problem does not occur.

**Version**
3.7 from firmware.ardupilot.org - I've tried several over the past few weeks, all show this problem.

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X quad

**Hardware type**
Bench tested with:
Holybro Kakute F7
FrSky R9 Mini or FrSky R-XSR receiver
Horus X10 or Taranis X9D with R9M module

**Logs**
Attached.
[simplemode.BIN.zip](https://github.com/ArduPilot/ardupilot/files/3046279/simplemode.BIN.zip)

khancyr, 2019-04-05:
hello,
thanks for the report. I think I have found the error

atshort, 2019-04-05:
Does the "NeedAnswer" label mean you need additional information from me?

atshort, 2019-09-25:
I have retested this on the latest dev build and can no longer reproduce the problem. I am closing.

closed 2019-09-25 by atshort