ArduPilot/ardupilot #11020: LOITER_* command upload fails if param4 = NaN

DonLakeFlyer, 2019-04-05:
The following loiter commands MAV_CMD_NAV_LOITER_UNLIM, MAV_CMD_NAV_LOITER_TURNS, MAV_CMD_NAV_LOITER_TIME need to be able to support a NaN in param4 in order to show that a heading is not specified in the command. The code here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mission/AP_Mission.cpp#L649 causes this to not work since it does not have these commands listed.

Related to https://github.com/mavlink/qgroundcontrol/issues/6783

WickedShell, 2019-04-05:
@DonLakeFlyer per the spec that's not appropriate. There is nothing in the spec that indicates those fields should ever have NaN's. They have specific values assigned for 0, 1, and all others specify an exit yaw angle.

DonLakeFlyer, 2019-04-05:
@WickedShell Per the spec, that spec doesn't make any sense. For other commands which specify a heading NaN means "don't change the heading. Without NaN in a heading change field there is no way to specify don't change to heading. I've already brought it up to the mavlink folks to update to docs.

Fixed Wing and Multi-Rotor use these fields differently as far as I can tell. Multi-Rotor uses it for heading which needs a NaN to specify "don't specify heading". Is that not correct? If so, how do you specify that I don't want to specify a heading?

ArduPilot is also failing mission uploads for things like MAV_CMD_IMAGE_STOP_CAPTURE and/or MAV_CMD_VIDEO_STOP_CAPTURE with have NaNs in param5. Not sure if it's both.

DonLakeFlyer, 2019-04-05:
I poked through the PX4 firmware code and for LOITER commands on a multi-rotor is treats this as a heading. Although as far as I can tell although it reads it out it doesn't yet implement heading changes for the command.

DonLakeFlyer, 2019-04-05:
> ArduPilot is also failing mission uploads for things like MAV_CMD_IMAGE_STOP_CAPTURE and/or MAV_CMD_VIDEO_STOP_CAPTURE with have NaNs in param5.

Although that is a less important issue since I don't think ArduPilot supports those commands yet. Even so, it should report unsupport command in that case.

WickedShell, 2019-04-05:
The rejecting NaN commands on our side is actually correct though. You are trying to represent new behavior without updating the spec, and if we hadn't NACK'd the command, you would actually be surprised at the result from the vehicle (as sending a NaN vs an angle would have an impact on vehicle behaviour, but wouldn't grant the one you are expecting).

The spec control over xtrack positions is actually being used by users at the moment, so I don't think that is something we can remove or backtrack on.

DonLakeFlyer, 2019-04-06:
Fixed wing treats the field differently with a x-track indicator and 0 or 1 is supported and NaN is not. For multi-rotor this field specifies a heading in radians. If I'm flying multi-rotor how then do you specify that you don't want to specify a heading?

DonLakeFlyer, 2019-04-06:
Also I am working to get the spec updated. The mavlink working group folks will discuss. I've already talked with them. To me this is not a spec change, this is simply incorrect/poor documentation of the existing spec. The concept of specifying a heading for multi-rotor and using NaN to specify no heading is a common thing throughout the mavlink commands. This one is just document poorly as far as I can tell.

WickedShell, 2019-04-06:
> For multi-rotor this field specifies a heading in radians

On a specific firmware sure, per the spec (which is what the implementations that have shipped were written against), this is not the case.

There are lots of bad things in the spec I don't disagree, but something like this is fundamentally a change, and is a breaking change to shipped implementations.

DonLakeFlyer, 2019-04-06:
I give up

closed 2019-04-06 by DonLakeFlyer

WickedShell, 2019-04-06:
I think the feature is not necessarily invalid, I just don't think it retrofits onto param4 the way it's proposed. (I also think there is value in letting fixed wing aircrafts specify their loiter exit angle which this change would preclude)

hamishwillee, 2019-05-02:
@WickedShell @auturgy Coming late to the party I can see now that the spec _has_ changed for the `LOITER_*` commands `param4` to use NaN - e.g. [MAV_CMD_NAV_LOITER_UNLIM.param4](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_UNLIM).
Don appears to have given up and just coded this as 0 for ArduPilot: https://github.com/mavlink/qgroundcontrol/pull/7356

I presume this was discussed within MAVLink while I was away. Can we work towards getting the ArduPilot behaviour changed to match the spec?

DonLakeFlyer, 2019-05-02:
> Don appears to have given up and just coded this as 0 for ArduPilot

That's a workaround that will be needed even if ArduPilot changes the NaN checking behavior.

> I also think there is value in letting fixed wing aircrafts specify their loiter exit angle which this change would preclude

It does not. The NaN is only for multi-rotor vehicles.

DonLakeFlyer, 2019-05-02:
> Coming late to the party I can see now that the spec _has_ changed for the `LOITER_*` commands `param4` to use NaN

I not sure that is the case. I think support for specifying a yaw for a multi-rotor vehicle was always there from the start. And if it was always there then you need some way to specify that you don't want to change the yaw. Just like the WAYPOINT command. What was wrong was that the spec was missing needed information.

hamishwillee, 2019-05-03:
> What was wrong was that the spec was missing needed information.

Strictly speaking the spec was ambiguous - it did not define the "don't change" case. But I don't see how else you can support this without using NaN, since any other value is a valid heading. The only other option is to either not support this parameter at all OR force the user to always specify a yaw - yuk!

Anyway, I think the spec is now correct. Let's see if @auturgy/@WickedShell agree.