ArduPilot/ardupilot #11031: Safety switch is not effective after software reboot on master

Jaaaky, 2019-04-07:
### Bug report

**Issue details**

_Using master, On Pixhawk1, after software reboot - reboot using MissionPlanner actions -, safety switch is can be switched on/off but servos are always off as if safety switch is on_

**Version**
_master arduplane_
_Works fine on plane3.9_
_Have done `git bisect` and found this as the 1st bad commit is f452d5c75236e980 which is a new I/O firmware - @tridge 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine
Didn't test other vehicles

**Airframe type**
_N/A_

**Hardware type**
_Pixhawk1_

Just build, flash firmware, activate safety switch - notice servos are active - reboot using GCS, reactivate safety switch - it gets active as light is stable but servos are free -

Jaaaky, 2019-04-10:
Follow-up;
The buggy behavior occurs with default params;
BRD_SAFETYOPTION,3
BRD_SAFETYENABLE,1

But if I set ```BRD_SAFETYENABLE,0``` it works fine.

tridge, 2019-04-20:
I have reproduced the issue, thanks!

PR #11158 (2019-04-20): Fixed safety state on soft reboot
fixes #11031 
thanks to @Jaaaky for reporting!
closed 2019-04-20 by tridge