ArduPilot/ardupilot #11048: Violent deceleration with DO_CHANGE_SPEED

Hasi123, 2019-04-08:
### Bug report

**Issue details**
During an auto mission, when a speed decrease with DO_CHANGE_SPEED is commanded, the copter violently decelerates causing a sudden and large tilt angle.
DO_CHANGE_SPEED should also respect the WP_ACCEL parameter.

Related issues: #8627 and maybe #9085

**Version**
Arducopter V3.6.7

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Octaquad

**Hardware type**
Pixhawk Cube (Black)

**Logs**
Log that shows the issue should be attached
[Uploading 2019-04-08 11-22-52.zip…]()

ChristopherOlson, 2019-04-08:
I have a fix for this coded up, but currently only for heli for the stopping distance, although the speed change in flight also works with multicopter

https://github.com/ChristopherOlson/ardupilot/commit/9c6725f64f95b77eb9c8534ba73dcee1090462cb

I spoke with @lthall this morning and he requested that I make a function and call it for both spline and regular waypoints and get a PR in for it. Just to let you know we are aware of the problem and working on it.

rmackay9, 2020-02-21:
This has been fixed in master and will probably go out with Copter-4.0.4 in a month or so.

Thanks for the report!

closed 2020-02-21 by rmackay9