ArduPilot/ardupilot #11055: Intel Aero RTF Building .px4

malfonsoNeoris, 2019-04-08:
#### Issue details
hi, i am trying to build the arducopter for the intel aero-rtf drone.
In the last version, i have found only the aero board, before there was the aero and aerofc-v1 boards.
Are this the same now ?
Furthermore, i have correctly build, following the waf configure --board=aero, and the waf copter commands
In the build/aero/bin folder, i have found a arducopter file.. but not the .px4 one.
how can i make the .px4 file ? or can i directly upload this file

thanks

#### Version
latest
#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Intel Aero RTF
#### Hardware type
Intel Aero RTF

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2019-06-11:
We no longer support PX4 builds in master so you won't see those.

Also, the aerofc board hasn't been ported to ChibiOS yet so we don't have support for it in master (as indicated in #8109). You'll need to use latest stable version.

closed 2019-06-11 by OXINARF

afarrukh-emhr23, 2021-06-13:
You can build .px4 for intel aerofc-v1 by switching to the last stable branch that supports this board: v3.6.12