ArduPilot/ardupilot #11056: Remove stale deployments

jaxxzer, 2019-04-08:
There is a stale PX4 folder here that should be removed:
http://firmware.ardupilot.org/Sub/beta/PX4/

jaxxzer, 2019-04-08:
@OXINARF says:
> It's debatable. If we remove them, people still using those targets won't be able to install them. We can remove it in beta, but we usually have beta have the latest released (beta or stable) version.

jaxxzer, 2019-04-08:
I think @OXINARF has a good argument to keep the folder, but I think this argument should only apply to the stable folder, not the beta folder.

Beta right now is 3.6.0 stable is 3.5.4. If I jump to a stable 3.5.5 without a beta, then I will not be able to update this PX4 target in the beta folder to point at the latest stable without updating all the (beta) targets to temporarily point at the stable firmware.

IamPete1, 2021-07-26:
Its gone

closed 2021-07-26 by IamPete1