ArduPilot/ardupilot #11060: the position using beacon fusion changes each time power on

Ncerzzk, 2019-04-09:
#### Issue details
i use a uwb device,4 beacons and 1 tag.
i set beacon position  at the initiation of beacon:
`	set_beacon_position(0,Vector3f(0,0,2.1));
	set_beacon_position(1,Vector3f(3.035,0.242,2.1));
	set_beacon_position(2,Vector3f(2.717,4.35,2.1));   
	set_beacon_position(3,Vector3f(0.3,4.35,2.1));
`
i have used EKF3 .

At the same position,the (x,y,z) changed each time i opened the vehicle.
For example,sometimes it's (2,3,0),next time  may be (4,1,0)

I'm puzzled and don't know what 's wrong.

However,the error of the position is OK.Position ranges  in 0.1m.(my uwb device'accuracy is 0.1m)

#### Version
_What version was the issue encountered with_
ArduRover V3.4.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ ✔] Rover
[  ] Submarine

#### Airframe type
Vehicle

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk

#### Logs
some logs in the mission planner
`EKF3 IMU0 initial beacon pos D offset = -1.9 (m)
EKF3 IMU0 initial pos NE = 1.5,-1.2 (m)
EKF3 IMU0 is using range beacons
EKF2 IMU0 initial beacon pos D offset = -2.0 (m)
EKF2 IMU0 initial pos NE = 2.1,-0.7 (m)
EKF2 IMU0 is using range beacons
EKF2 IMU0 tilt alignment complete
Failsafe trigger 0x1
PX4v2 00250030 32365115 37353639
PX4: 1e7ed30b NuttX: 1472b16c
ArduRover V3.4.2-dev (f7f4b436)
EKF3 IMU0 tilt alignment complete
EKF3 IMU0 initial yaw alignment complete
EKF3 IMU0 initialised`

khancyr, 2019-04-09:
hello, 

that isn't an error.
On start, the EKF will use the first 100 first reading to estimate a local center for the beacon system.

Ncerzzk, 2019-04-09:
> hello,
> 
> that isn't an error.
> On start, the EKF will use the first 100 first reading to estimate a local center for the beacon system.

Hello,thanks for reply!
I'm sorry that i'm new to ArduPilot. Maybe i didn't describe it clear.

I put the vehicle at a same place,while its position changes when i shutdown and restart.

If there is nothing wrong,that means  i couldn't get the absolute position using uwb?

Ncerzzk, 2019-04-09:
i try to set a const value instead of 
`           receiverPos.x = rngBcnPosSum.x * tempVar;
            receiverPos.y = rngBcnPosSum.y * tempVar;
`
while it doesn't work,and the position err is larger.

Ncerzzk, 2019-04-16:
I have solved the problem.
if need the absolute position using uwb,
sub the bcnPosOffsetNED.x and bcnPosOffsetNED.y at last

closed 2019-04-16 by Ncerzzk

imnobodyet, 2019-07-31:
@Ncerzzk could you please tell me where to do the sub ?