ArduPilot/ardupilot #11081: Copter:WP_YAW_BEHAVIOR not effecting

thomasonegd, 2019-04-11:
### Bug report
Copter:WP_YAW_BEHAVIOR not effecting
**Issue details**
I'm running Copter 3.7.0(SITL)，the WP_YAW_BEHAVIOR is set to 1(or 2)，but the pilot never change it's heading during all the WPs,the heading keeps the value like it takes off.
**Version**
3.7.0

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QUAD - X
**Hardware type**
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_
[my_waypoints.txt](https://github.com/ArduPilot/ardupilot/files/3067786/my_waypoints.txt)

amilcarlucas, 2019-04-23:
Please retest with the latest master, this might have been fixed today.

amilcarlucas, 2019-08-27:
I tested it, I could no reproduce this.

rmackay9, 2019-08-28:
ok, thanks.  I will close this.

closed 2019-08-28 by rmackay9