ArduPilot/ardupilot #11084: Adding FrSky Telemetry downlink

yanivasy, 2019-04-11:
Currently there are 3 commonly used telemetry protocols: mavlink (mode 1), native Frsky (mode 4) & FrSky passthrough (mode 10).
The usage of all requires both uplink and downlink: uplink for requesting data while downlink for receiving the answers.
By implementing another telemetry mode, that just send the sensors data without being queried, all telemetry data can be displayed easily upon OpenTx transmitter 
The suggested telemetry mode could be the well know FrSky passthrough which have many supporting scripts such as the fantastic @yapuu.
This way, any datalink supporting 57600 baud rate can be used for telemetry whereas the conversion to SPORT can be done by simple hardware within the transmitter.     
Furthermore, the remine unused uplink can be used for other usages (controling, extra channels etc.).
This issue came out in the following thread by @schwabe
https://github.com/opentx/opentx/issues/6388

IamPete1, 2019-04-11:
Mavlink just streams data with no request. I'm not sure I understand what the need is? FrSky pass-through works very well without this.

Or are you suggesting that the Frsky protocol could be sent over another telemetry link, such as SIC Radios and piped directly into OpenTx? But again not really any advantage to this.

schwabe, 2019-04-11:
I am from OpenTX here. He wants to connect an FC directly to an OpenTX remote's sport and it to work.

IamPete1, 2019-04-11:
ah OK, so because OpenTX does not request any messages none are sent. Usually the messages would be requested by the ground side RF module, but in this case that is not present. 

So the solution would be to just stream the messages without any requests.

I guess we could do that, but personally if i had a 'full' telem link available i would connect it to a proper ground station, even if it is just QGC on a phone using OTG.

yanivasy, 2019-04-11:
@IamPete1,
I'm using another  full telemetry channel for Mission Planner upon a laptop by high 1w datalink for both planning and supervising.
Taranis OpenTx baesd transmitter includes a  build-in telemetry which is easy to carry.
However, Taranis build-in transmitter limits my range and therefore a more powerful transmitter (SiK for example) is required to be added externally and as a consequences eliminates its build-in telemetry.
By just stream the messages without any requests its build-in telemetry can be used easily (by adding  a minor inverter).

kilrah, 2019-04-11:
> I'm using another full telemetry channel for Mission Planner upon a laptop by high 1w datalink for both planning and supervising.

Then you could be using the same for display on the radio, with the converter you were pointed to already...

https://github.com/zs6buj/MavlinkToPassthru

IamPete1, 2019-04-11:
looks like this would not be too hard, relevant code is here
https://github.com/ArduPilot/ardupilot/blob/49b0a001824a0e7b22d77e3578aa944f08c7ca7b/libraries/AP_Frsky_Telem/AP_Frsky_Telem.cpp#L98
You could probably just comment out that if and see if it works.

yanivasy, 2019-04-11:
Thanks @IamPete1,
I'm trying to compile a copter version for Pixhawk2 cube black after disabling the "if" case (comment on line 98 & line 169) but it seems some libraries are missing. Please compile a trial version for me to test until I'll arrange all missing libraries.

yanivasy, 2019-04-11:
@kilrah,
I need the transmitter to be separated completely from ground station.
Using the same telemetry for both telemetry & transmitter constrain this freedom.   
The MavlinkToPassthru implemented by Teensy 3.2 doesn't free-up the uplink channel as @schwabe suggestion.

kilrah, 2019-04-12:
The converter can be connected to your existing downlink in Relay mode, and your X8R connected to it to to wirelessly transmit to the radio.

the-nicolas, 2020-06-09:
Any update on this?

IamPete1, 2020-06-09:
@the-nicolas I'm still struggling to understand the use case to be honest

yanivasy, 2020-07-16:
> @the-nicolas I'm still struggling to understand the use case to be honest

The use case is very simple:
If you are using only a radio transmitter (without a MavLink protocol ground station) then currently you can get telemetry data by FrSky passthrough (Mode 10) directly to the a Taranis radio.
When you extend the Taranis range by adding RFdesign RFD900 SIK or any other low cost high power data link, you are able to control the FC by PPM signal injected directly from Taranis external bay connector to the RFD900.
However, telemetry data is not available (although you have both up-link and down-link channels) since the FC in FrSky passthrough mode, doesn't streams data with no request.
If the data will be stream (simply by removing the "if" sentence), telemetry will be available and also the up-link channel.
This, without any additional hardware (as the suggested Tennsy 3.2) beside a simple diode for SPORT conversion.

IamPete1, 2020-07-16:
> If the data will be stream (simply by removing the "if" sentence), telemetry will be available and also the up-link channel.
@yanivasy have you tested this? I can compile a test code if you tell me which flight controller.

Are you aware that the TXMod will now do FRSky telm for you? You send it mavlink and it feeds the telem steam on to OpenTX.
https://files.rfdesign.com.au/Files/documents/TXMOD%20-%20Adding%20S.PORT%20Support.pdf

yanivasy, 2020-07-16:
> @yanivasy have you tested this? I can compile a test code if you tell me which flight controller.
>
> Are you aware that the TXMod will now do FRSky telm for you? You send it mavlink and it feeds the [...]

I haven't tried to remove the "if" sentence, I'll appreciate a compile est code for Cube black

I wasn't aware of the TxMod S.Port upgrade - nice feature.
This however, requires a TxModule along with RFD900 (approximately an addition of 100$).

IamPete1, 2020-07-16:
@yanivasy [One Copter CubeBlack](https://drive.google.com/file/d/1NqXTCk-rAsDAe64zqJ4P3oSIt--SilkT/view?usp=sharing)

No promises it will even run, I would not advise flying it.

yanivasy, 2020-07-25:
Thanks @IamPete1 ,
I've began testing it, and now data streams out of the serial port Cube without any FrSky receiver connected.
However, while connecting the serial port thru an RS232 driver (to invert the signal) to S.PORT of Taranis RC, the data is not decoded by Taranis RC.
I'll trying to find out why and report any insights.