ArduPilot/ardupilot #11089: Build failed in [493/495] apj_tool build/CubeBlack/bin/ardurover Typeerror

DaHaiHuha, 2019-04-11:
### Bug report

**Issue details**

When build rover in macos mojave with python2.7, the build failed in 493/495 with TypeError:
```
Build failed
Traceback (most recent call last):
  File "/Users/apple/workspace/ardupilot/modules/waf/waflib/Task.py", line 338, in process
    ret = self.run()
  File "Tools/ardupilotwaf/chibios.py", line 78, in run
    if defaults.find():
  File "/Users/apple/workspace/ardupilot/Tools/scripts/apj_tool.py", line 93, in find
    i = self.firmware[self.offset:].find(magic_str)
TypeError: argument should be integer or bytes-like object, not 'str'
```

**Version**
[master- 49b0a00](https://github.com/ArduPilot/ardupilot/commit/49b0a001824a0e7b22d77e3578aa944f08c7ca7b)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ * ] Rover
[  ] Submarine

**Logs**
```
[493/495] apj_tool build/CubeBlack/bin/ardurover
Loaded binary file of length 2294780
Waf: Leaving directory `/Users/apple/workspace/ardupilot/build/CubeBlack'
Build failed
Traceback (most recent call last):
  File "/Users/apple/workspace/ardupilot/modules/waf/waflib/Task.py", line 338, in process
    ret = self.run()
  File "Tools/ardupilotwaf/chibios.py", line 78, in run
    if defaults.find():
  File "/Users/apple/workspace/ardupilot/Tools/scripts/apj_tool.py", line 93, in find
    i = self.firmware[self.offset:].find(magic_str)
TypeError: argument should be integer or bytes-like object, not 'str'
```
The configuration information is as below:
```
➜  ardupilot git:(master) ./waf configure --board CubeBlack --python=python2
Setting top to                           : /Users/apple/workspace/ardupilot 
Setting out to                           : /Users/apple/workspace/ardupilot/build 
Autoconfiguration                        : enabled 
GCOV code coverage analysis              : no 
Setting board to                         : CubeBlack 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/local/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/local/bin/arm-none-eabi-gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Checking for program 'make'              : /usr/bin/make 
Checking for program 'arm-none-eabi-objcopy' : /usr/local/bin/arm-none-eabi-objcopy 
Including /Users/apple/workspace/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/fmuv3/hwdef.dat
Adding environment OPTIMIZE -O3
Removing PB0
Removing PB1
Setup for MCU STM32F427xx
Writing hwdef setup in /Users/apple/workspace/ardupilot/build/CubeBlack/hwdef.h
Writing DMA map
Generating ldscript.ld
Adding defaults.parm
Checking for env.py
env set OPTIMIZE=-O3
env set HAL_WITH_UAVCAN=1
env set PROCESS_STACK=0x2000
env set MAIN_STACK=0x400
env set IOMCU_FW=0
env set CPU_FLAGS=['-mcpu=cortex-m4', '-mfpu=fpv4-sp-d16', '-mfloat-abi=hard', '-u_printf_float']
env set APJ_BOARD_ID=9
env set APJ_BOARD_TYPE=STM32F427xx
env set FLASH_RESERVE_START_KB=16
env set DEFAULT_PARAMETERS=/Users/apple/workspace/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/CubeBlack/defaults.parm
env set CHIBIOS_BUILD_FLAGS=USE_FATFS=yes CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f4xx.mk CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F4xx/platform.mk MCU=cortex-m4 ENV_UDEFS=-DCHPRINTF_USE_FLOAT=1
Enabling ChibiOS asserts                     : no 
Checking for intelhex module:                : disabled 
Checking for HAVE_CMATH_ISFINITE             : yes 
Checking for HAVE_CMATH_ISINF                : yes 
Checking for HAVE_CMATH_ISNAN                : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : not found 
Checking for header byteswap.h                 : not found 
Checking for HAVE_MEMRCHR                      : no 
Checking for program 'python'                  : python2 
Checking for python version >= 2.7.0           : 2.7.16 
Checking for program 'python'                  : python2 
Checking for python version >= 2.7.0           : 2.7.16 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Gtest                                          : STM32 boards currently don't support compiling gtest 
Checking for program 'arm-none-eabi-size'      : /usr/local/bin/arm-none-eabi-size 
Benchmarks                                     : disabled 
Unit tests                                     : disabled 
Scripting                                      : disabled 
Scripting runtime checks                       : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
```

DaHaiHuha, 2019-04-15:
```
i = self.firmware[self.offset:].find(magic_str)
TypeError: argument should be integer or bytes-like object, not 'str'

```
This occurs unexpectedly and I didn't change anything from the original code, anybody could help?

peterbarker, 2019-04-17:
On Mon, 15 Apr 2019, ChengDahai wrote:

> i = self.firmware[self.offset:].find(magic_str)
> TypeError: argument should be integer or bytes-like object, not 'str'
> 
> This occurs unexpectedly and I didn't change anything from the original code, anybody could help?

Steps to reproduce?  Which python is being used on your system?

DaHaiHuha, 2019-04-17:
> Steps to reproduce? Which python is being used on your system?
I configured python2.7/3.5/3.6 to build the code but still I met the same error. When I use python3.7 to build, another error as described in  [issue #10333](https://github.com/ArduPilot/ardupilot/issues/10333)

closed 2019-04-18 by peterbarker

peterbarker, 2019-04-18:
@DaHaiHuha Thanks for the report.  Should be fixed in the master branch now.

PR #11142 (2019-04-18): Tools: apj_tool.py: python3 fixes
Closes #11089

Marking this as WIP as I haven't managed to test it yet.

On master, any parameters in defaults.parm for CUAVv5 (the autopilot I have handy) causes the thing to lock up at boot.