ArduPilot/ardupilot #11090: Hidden (Major) dangers in using Do_land_start with any Quadplane

pompecukor, 2019-04-11:
### Bug report

**Issue details**

We realized the following undesired and very risky situation with current code during a random training.

We are (I am guessing the few, as no one seemed to have noticed this simple problem) who prefer full (preset) control of of VTOL landing, especially with regards to landing approach and so on. So we never rely on QRTL and/or RTL land. Rather we use VTOL land in every mission with approach and all. Then for RTL we have a Do_land_start somewhere, a few waypoints from the VTOL land. For this always have RTL_AUTOLAND set to: 2 (which makes the mission pointer jump to whatever comes right after the Do_land_start "marker". 

This is a (best) practice that we carried over from Traditional fixed wing. So in case of and failsafe and thus RTL. The Unit would fly to the landing approach that we set with the market and proceed from there.

I know there is a wonderful new feature of master that lets landing be almost automated in case of RTL with the auto-loiter to alt then transition into the wind. However we found this approach too risky for our clients as it require above average situational awareness that even a Mobile phone tower in the pack if the loiter to alt could cause a crash. Same goes from when flying from a point with elevation at the back, like the side of a hill or next to a very tall building. We fund it safer and more straightforward, to always just have a predefined approach and landing. Yes the other is much simpler, but there is a risk as described and only needs one time to forget that the plane would make a big circled before landing. The only place I would advise to use that feature is if the user is always flying in more or less flat terrain and never next to any object taller than 35m.

Anyway back to the topic. The above approach works well as usually. Here is a simple mission example of the safest way that I know to operate a VTOL in the current code (till we get some smart RTL features in the mix).

[image]

As you can see in this way, you can go as far and as high as you want (with for example Guided mode), then proceeding with auto or an RTL/failsafe at anytime will return you to WP6 at a safe asltitude (which you have set in RTL_ALT parameter) and then proceed to loiter to a more suitable altitude for transition on WP7 (loiter to alt), Then approach vector set with WP8, with also initiates transition and finally VTOL land on the set point.

So again, this is fine and dandy. However today we realized that there was a huge rist to this with Quadplanes. It at anytime during the mission a failsafe occurs, then the UAV would go directly to WP6. For traditional fixed wing this is not really and issue. However with Quadplane this should NOT happened anytime during VTOL mode.
Unfortunately it does.

We were out with a client practicing a bit of Qloiter after some missions as, we still had some energy left on the last pack. Suddenly, the UAV started to climb on its own. Immediately I realized what happened and snatched the RC from the client, but if they were alone it would have been another story. Low voltage Fail-safe kicked in and the VTOL wanted to climb back to transition altitude and transition then fly to WP 6 and redo everything from that. We would have crashed as the battery might have not been enough for a full up and down transition.  

LONG STORY SHORT. Anyone is free to test this in SITL sim. RTL and the commanded Do_land-start gets called even during VTOL phases, expecially scary is VTOL_LAND, VTOL_TAKEOFF, Qstab and Qloiter. I did not test Qland.

**Fails safe should never initiate in those modes with Do_land_start.**

I don't think it needs a big imagination, but I will give a sever example: having any sort of Low voltage FS set to a very low level. Imagine the UAV returning from a long mission with a mission like the above. And very low battery, but still safe. Then just during the VTOL land phase the low voltage FS kicks in, **that plane is doomed!**.

**And it gets worse!** I tested in SITL. The biggest danger is not even a depleted battery, but rather anysort of failsafe. Say for example a radio failsafe for any ready. As long as RTL is activated.
Try this in the sim. Hit auto in the above mission, and as soon as the VTOL starts to climb, lets say at 3meters. Hit RTL. See what happens? **It starts to transition at that 3 meters crashing into any and everything in its way.** Same goes for Landing phase, only in that case a low battery only adds to the "excitement".

I rest my case.

**Version**
Lastest master and obvious all of the version before will be the same.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

pompecukor, 2019-04-11:
Obvious solutions:

RTL should ignore any Do_land_start markers if the UAV is in any of the following modes or startes:
VTOL land, VTOL Takeoff,  Qloiter, Qstab, Qland.

I am not sure about Qhover.

Obvious question is what should it do in those cases. My most sensible answer is VTOL_Land or Qland. if in any of the autonomous mode: VTOL land and VTOL takeoff. The reason is simply that the VTOL is in a good (horizontal) position in these bases.
As for semi autonomous modes, Qloiter, Qstab. That is a bit more tricky to answer. But I would think VTol hover to home and then Qland would make send is most cases. Qland can be more dangerous if you just lost a radio link.

DavidIngraham, 2019-04-13:
Hi @pompecukor,
Some of these behaviors for battery failsafes have actually changed recently. The code that is responsible for selecting an action based on a failsafe is located here: https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/events.cpp

You can see how it's changed over time by selecting the tag for different releases in the branch dropdown.

Since the introduction of quadplane, RC and GCS failsafes will trigger a QLAND if the vehicle is in a quadplane mode. However, it does not catch the case where a quadplane is flying as a vtol in auto.

In 3.7 and 3.8, battery failsafes would trigger QLAND if the vehicle was in an auto mode (but not in VTOL Auto, same as above). Since 3.9, a battery failsafe will only trigger a QLAND if that is the selected battery failsafe option. I support the idea of adding an additional battery failsafe option that is "RTL if in Fixed wing Mode, QLAND if in VTOL Mode"

@WickedShell , do you have thoughts on this?

pompecukor, 2019-04-13:
Thanks for the tag David.

jace26, 2019-04-15:
I definitely agree to this line of thought. The current handling of failsafes in Auto while multicopter (MC) isn't safe. The option to land if certain criterion are met (battery, location, distance to home/wp) or RTL MC (under certain conditions) or RTL fixed wing (under certain conditions) would be ideal. 

We had a similar problem and considered using the advanced failsafe options which allow you to jump to a WP in the plan (after all your other wps) which is handy because you can have it come back and land using a predefined approach like you said @pompecukor , but getting it to handle doing this from a fixed wing or VTOL config while in "Auto" is difficult. One way to fix this that might be universally good is to have the FC ignore VTOL_takeoff commands when in fixed wing. This would allow you to place a VTOL_Takeoff as your initial response to a Failsafe condition, and if the aircraft is fixed wing it will skip it and go to the predefined wp, if the aircraft is in MC mode it will climb out and transition. Right now, this can't work because (in SITL) when you execute a VTOL_Takeoff mode in fixed wing it will first transition to VTOL, then execute the takeoff, and finally retransition to fixed wing (if the next wp is far away). Obviously, this is a ridiculous response if you are in a failsafe condition, and I actually cannot think of any good reason to not ignore a VTOL_Takeoff command while fixed wing in auto. 

another thing to consider in the future will be failsafe conditions based on ESC/motor feedback (i.e. a dying Fixed wing power section demands a MC failsafe response, and visa-versa)

pompecukor, 2019-06-24:
I would say it is very rear that user would be in any Qmode outside of close proximity. As in case of 99% of vtols. it is only used for take-off and landing. Also in other cases (flying Qloiter or Qstab or Qhover) it is typically in LOS, so again would be close to home.

So a very simple and basic solution would already affect 99% of cases in a positive way. If we simply had **any sort of Failsafe while inside of RTL Radius ignore any Do_land start and ALWAYS go to into QRTL** (hover to home and land there). I cannot imagine any risk if this was implemented.

ovolkmann, 2019-06-24:
@pompecukor , this does not help those of us who are landing at a location other than Home. We need a solution that would avoid RTL/QRTL during a landing phase of an automatic waypoint mission. The autopilot should recognize that it is in a DO_LAND_START sequence and complete the landing regardless of any failsafe events

pompecukor, 2019-06-24:
Yes, you are the 1% that I spoke about....
As devs are very busy, I am recommending a simple to implement fix that will help 99% of users. And will not matter for the 1%. As this works only if you are in RTL radius. Then on the long run, with time, the greater scope may be handled.

HIT: _Maybe you can make donation to autopilot.org and become a corporate partner and find a dev that will spend time with you on the more complicated solution for the remote landing cases. Your business is making money of of this platform, it is only right that you show some support._

ovolkmann, 2019-06-24:
@pompecukor , I am a corporate partner :)

pompecukor, 2019-06-24:
Then forget I mentioned it. Welldone.

Naterater, 2019-12-04:
Any chance this could be fixed soon?