ArduPilot/ardupilot #11114: Vtol configuration Tailsitter alway not armed

malpoma, 2019-04-14:
Please check the always arming mode for vtol configuration.

if i put 0 to all parameter of arming, only the wing configuration is armed not the vtol...and in this case the pwm motor don't work.

if put arming rudder 1 or 2 the pwm motor on vtol conf start work...

many thx

IamPete1, 2019-04-14:
log can be found here https://discuss.ardupilot.org/t/tailsitter-problem-at-pwm-motor/41085

to summarise ARMING_REQUIRE 0 does not arm VTOL motors but plane stuff works as it should. Tested on a tailsitter.

kd0aij, 2019-12-21:
@IamPete1 Has this been fixed?

IamPete1, 2019-12-22:
@kd0aij Not that I know of

kd0aij, 2019-12-22:
this may be related to the fact that a copter state machine must be running for the motors to spool up, but plane isn't calling it in non-VTOL modes.

IamPete1, 2021-08-24:
closing in favour of https://github.com/ArduPilot/ardupilot/issues/9676

closed 2021-08-24 by IamPete1