ArduPilot/ardupilot #11117: Rover time rollover issue

peterbarker, 2019-04-15:
### Bug report

Time roll-over seems to cause Rover significant issues, including reboot

**Version**

```
2019-04-10 07:58:25.91: STATUSTEXT {severity : 6, text : Mission Planner 1.3.64}
2019-04-10 07:58:27.60: STATUSTEXT {severity : 6, text : ArduRover V3.5.0-rc2 (90c9ec29)}
2019-04-10 07:58:27.60: STATUSTEXT {severity : 6, text : ChibiOS: 35ad8a79}
2019-04-10 07:58:27.62: STATUSTEXT {severity : 6, text : MatekF405 0027001B 4D4B5006 20343252}
```

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Airframe type**
boat

**Hardware type**
Matek F405 Wing

**Logs**

TBA

[image]

[image]

tridge, 2019-04-15:
temperature increased over 4hrs leading up to reboot:
[image]

peterbarker, 2019-04-15:
[image]

rmackay9, 2019-04-15:
I guess we've concluded that this is more likely a heat or power issue so I think I'll close this one.

closed 2019-04-15 by rmackay9