ArduPilot/ardupilot #11119: AHRS/EKF: get_vert_pos_rate always returns zero

rmackay9, 2019-04-15:
I've found that the AP_AHRS_NavEKF::get_vert_pos_rate() function always returns zero.

Below is a screen shot of some code I added to SITL to output the EKF's normal climb rate (up is negative), the baro climb rate (up is positive) and the output from the get_vert_pos_rate function.
[image]

The result is that the pos_rate is always zero.
[image]

WickedShell, 2019-04-15:
In EKF2 `posDownDerivative` is only set during the `StoreOutputReset` which is only called during the filter initialization. (`posDownDerivative` is the basis for the call to get_vert_pos_rate)

priseborough, 2019-04-15:
The derivative is supposed to calculated by the output observer - I will check the code.

priseborough, 2019-04-15:
I've checked the code and the posDownDerivative calculation which is supposed to be done inside the output observer is missing. 

It was added originally with this commit https://github.com/ArduPilot/ardupilot/commit/52fd369b2f82c2eb879fb13ac39cc04dbb8ca5fc

And removed when the output observer was re-written here: https://github.com/ArduPilot/ardupilot/commit/6523481c76d454f0fe4ec302a8f771bf3b1363da

The method we should be using is as used by the  PX4 ecl library calculation - see https://github.com/PX4/ecl/blob/master/EKF/ekf.cpp#L617

rmackay9, 2019-04-16:
@priseborough, do you think you could fix this?  I could give it a try but I don't have a lot of confidence in my abilities to get it right.  I'm certainly happy to help test it.

rmackay9, 2019-09-17:
@priseborough, I had a quick try and getting this to work but I don't think I've got it quite right. https://github.com/rmackay9/rmackay9-ardupilot/commit/41d236a34fcb4502850f0577caaa04ab1c47abb7

Any chance you could have a look at this?  It's quite important for the Copter-4.0.0 release which I hope we can enter beta testing within the next couple of weeks.

priseborough, 2019-09-21:
@rmackay9 I've created a WIP branch here for the EKF2 changes https://github.com/priseborough/ardupilot/tree/ekf2VertPosDeriv-wip

priseborough, 2019-09-21:
I'm in the process fo testing and tuning it and still need to update the comments to make the distinction between the vertical channel complementary filter used for the height derivative calculation and the one used for the normal NED state outputs.

priseborough, 2019-09-21:
The output tracks, but it is 'over compensating' for changes in climb rate so further tuning is needed. The following shows NKF1.dPD which is the vertical position derivative data. This was generated using SITL Copter.

[image]

priseborough, 2019-09-21:
I will also compare it against the results obtained using the original method from https://github.com/ArduPilot/ardupilot/commit/52fd369b2f82c2eb879fb13ac39cc04dbb8ca5fc  . The output observer has changed structure since then, but the method should still be valid.

priseborough, 2019-09-22:
I tried a different approach to what is used on the PX4/ecl library, this time using the same threes state complementary filter that ArduPlane TECS used for the heigh channel prior to the EKF.

See https://github.com/priseborough/ardupilot/tree/ekf2VertPosDeriv2-wip

The tracking is better and it won't be affected by EKF velocity state bias errors.

[image]

priseborough, 2019-09-22:
We may want to make the crossover frequency tuneable and I need to do another pass to make sure it's states are always reset whenever the EKF vertical position or velocity state is reset.

rmackay9, 2020-04-21:
The vibration failsafe including EKF changes have gone into master so we can close this now, thanks!

closed 2020-04-21 by rmackay9