ArduPilot/ardupilot #11131: Dev Env: enable travis on lthall's repos

rmackay9, 2019-04-17:
Some of the core devs have travis (CI?) enabled on their repos.  This allows automatic testing of their repos after every commit.  @lthall is doing a lot of code changes these days and it would be helpful if we could get this turned on for him as well.

OXINARF, 2019-04-17:
This is something each user can do on its own. It's a bit late over here but I'll try to write some instructions tomorrow so it can be documented.

rmackay9, 2019-04-17:
@OXINARF, thanks!

lthall, 2019-04-17:
Thanks mate!!

OXINARF, 2019-04-18:
Alright, so I actually helped @lthall do this last night, which was great for checking the steps.

Tutorial **for new users**:

1. Go to https://travis-ci.com and click the big _Sign up with GitHub_, follow the instructions to allow Travis to use your GitHub login.
2. Once you are back in Travis, click the Authorize button so that Travis has access to your repositories. You then have two options:
   - allow Travis have access to all your repositories
   - select which repositories you want Travis to have access to

   First option is the easiest one, but both work.
3. You are done, you now just have to push new commits into your branches for them to appear on Travis and start building.

I'm closing this now, but anyway having doubts can write them here or ask in 
Gitter (http://gitter.im/ArduPilot/ardupilot). I've also changed the title so it is easily searchable.

closed 2019-04-18 by OXINARF