ArduPilot/ardupilot #11136: syslog, kern.log, messages are filled

blueForestIcarus, 2019-04-17:
running copter 3.7 dev:

Ardupilot seems to be filling up the disk with log messages in /var/log/kern.log /var/log/syslog and /var/log/messages.

Most of the messages are like the following.
```
[  481.993933] rcio spi1.0: rcio_pwm: requested update on 1 to 50
[  481.994056] rcio spi1.0: rcio_pwm: requested update on 2 to 50
[  481.996450] rcio spi1.0: rcio_pwm: requested update on 0 to 490
[  481.996592] rcio spi1.0: rcio_pwm: requested update on 0 to 490
```

About hundred of theses are logged every second and over the course of a couple hours it fills a 32 gigabyte sd card.

rmackay9, 2019-04-17:
Which flight controller?  I guess one of the linux boards?

blueForestIcarus, 2019-04-18:
It is a raspberry pi 3 with navio2.

Opening up mavproxy I notice that I get a message:

> Critical: AP_Logger: stuck thread ()

q-wertz, 2020-05-29:
My syslog also gets quite big with non human readable stuff...

```
May 29 13:45:24 navio sh[436]: �G��
May 29 13:45:24 navio sh[436]: �"
May 29 13:45:24 navio sh[436]: #M�#030#015#001#001��}#001����<��#002?
May 29 13:45:24 navio sh[436]: ���(#016#001#001��}#001����<��#002?
May 29 13:45:25 navio sh[436]: ���#034#017#001#001#036k�
May 29 13:45:25 navio sh[436]: �#034��#A_�E!#034?�S��̶X�P�
May 29 13:45:25 navio sh[436]: =�6�#024#020#001#001J
May 29 13:45:25 navio sh[436]: R���
May 29 13:45:25 navio sh[436]: �"
May 29 13:45:25 navio sh[436]: �0�#030#021#001#001�Q����e�<��#002?
May 29 13:45:25 navio sh[436]: L��(#022#001#001�Q����e�<��#002?
May 29 13:45:25 navio sh[436]: f��3#023#001#001�#002AP_Logger: stuck thread ()
May 29 13:45:25 navio sh[436]: ���#011#024#001#001
May 29 13:45:25 navio sh[436]: #003A#004#003�#032�#034#025#001#001![�
May 29 13:45:25 navio sh[436]: ��������
May 29 13:45:25 navio sh[436]: �#015�F�#037#026#001#001#001/� #003/�
May 29 13:45:25 navio sh[436]: #017�#020
May 29 13:45:25 navio sh[436]: #020
May 29 13:45:25 navio sh[436]: ��
May 29 13:45:25 navio sh[436]: ����#006#027#001#001};#023
```