ArduPilot/ardupilot #11137: Attitude controller can use data from an unhealthy gyro under certain circumstances

jschall, 2019-04-17:
### Bug report

**Issue details**
Copter (and presumably other vehicle types) use AP_AHRS::get_gyro() to retrieve filtered angular velocities that have been corrected for estimated gyro bias. Under certain circumstances, this function can return data from an unhealthy gyro.

When the IMU associated with an EKF core has a failure, that EKF core will start using data from a different IMU. An IMU failure will not trigger the primary EKF core to change. AP_AHRS_NavEKF::get_gyro() returns data from the gyro that the EKF core is configured to use, not the one that it is currently using.

Several fixes are needed:
1. AP_AHRS_NavEKF::get_gyro() should be guaranteed to return data from a healthy gyro if one exists.
2. An EKF primary core switch should be triggered when the IMU associated with the primary EKF core becomes unhealthy.
3. When an EKF core switches to a different IMU because the current IMU associated with that core becomes unhealthy, the IMU bias states should be reset.

**Version**
Copter 3.6.7

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Hex

**Hardware type**
Cube Black

**Logs**
https://discuss.cubepilot.org/t/cube-black-crashed-ac-3-6-7-firmware-hexacopter/394/3

peterbarker, 2019-04-17:
Can this be reproduced in SITL / autotest?

rmackay9, 2019-04-17:
Here's a related issue: https://github.com/ArduPilot/ardupilot/issues/7595

jschall, 2019-04-18:
@peterbarker No, we need a way to fail a gyro in SITL. It needs to be marked unhealthy and output bad data from ins.get_gyro().