ArduPilot/ardupilot #11169: (Incorrect) Use of GPS_OK_FIX_3D to determine GPS Health

sunilchomal, 2019-04-23:
### Bug report

In:
https://github.com/ArduPilot/ardupilot/blob/d5d81f3a1cf12f8b97978b704b86d503d03fdcf5/ArduPlane/GCS_Plane.cpp#L108

GPS_OK_FIX_3D is being used to determine GPS health, which I think is not correct

Instead of :

```
if (gps.status() >= AP_GPS::GPS_OK_FIX_3D && gps.is_healthy()) {
    control_sensors_health |= MAV_SYS_STATUS_SENSOR_GPS;
}
```

I suggest it should be:

```
if (gps.is_healthy()) {
    control_sensors_health |= MAV_SYS_STATUS_SENSOR_GPS;
}
```

This would make this code inline with code for GPS health in other vehicles such as Rover, Copter etc.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2019-04-23:
I agree actually.  I think to be consistent with other sensors we should say that "healthy" means the sensor is responding with reasonable numbers.  For example, we don't say that a range finder is unhealthy when it's ranges are below RNGFND_MIN_CM.

amilcarlucas, 2019-04-23:
@sunilchomal can you submit a PR?

sunilchomal, 2019-04-24:
@amilcarlucas PR #11178 submitted.

WickedShell, 2020-07-17:
Closing as we currently actually do strongly want this in plane as it is, and the PR to change it was rejected as undesired change.

closed 2020-07-17 by WickedShell