ArduPilot/ardupilot #11170: Plane: Geofence Altitude Runaway

mareschn, 2019-04-23:
### Bug report

**Unique Parameters**
RTL_AUTOLAND= 2 (landing sequence)
FENCE_ACTION = 4 (RTL)

For a fence breach we wanted to have the plane go to an auto-land sequence.  This works as expected to breach the walls of the fence but the ceiling causes problems.  Instead of going to the landing sequence in AUTO mode, the plane stays in RTL mode, loiters around the next waypoint after the DO_LAND_START, and increases in altitude forever. 

I uploaded a simulation .tlog from the beta/dev version, but all recent versions seem to do this.  See the start of the log for an example of proper execution, some wall breaches in the middle, and finally a ceiling breach showing the problem.

**Version**
3.97

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Standard Airplane

**Hardware type**
Pixhawk

**Logs**
[sim.zip](https://github.com/ArduPilot/ardupilot/files/3108599/sim.zip)

mareschn, 2019-04-23:
I can also confirm that the opposite happens when the geofence floor is breached: the aircraft goes into RTL and descends until ground contact.

IamPete1, 2022-08-05:
Should be fixed by https://github.com/ArduPilot/ardupilot/pull/21152