ArduPilot/ardupilot #11180: Plane - Enhanced RTL algorithm

MFlghts, 2019-04-24:
### Feature request

**Is your feature request related to a problem? Please describe.**

Problem with RTL can happen as most of the obstacles located aside of the flight pass, but not in the front usually! RTL starts to turn immediately towards home and can lead a plane into an obstacle, This happened to me couple of times.

**Describe the solution you'd like**

The solution is to first climb up to desired RTL alt while keeping current direction (straight forward), and after reaching RTL Alt start to turn home.

**Describe alternatives you've considered**
Can be made as an option.

**Platform**
[  ] Plane

magicrub, 2020-05-18:
seems reasonable

Hwurzburg, 2020-05-18:
Already PR posted to solve this #14371 ....tested and works well...

closed 2020-05-18 by tridge