ArduPilot/ardupilot #11185: Cant load NuttX once loaded Chibios

FLYBYME, 2019-04-24:
### Bug report

**Issue details**

I have installed the latest copter/plane/rover (Chibios) and the board boots fine. I try to load the latest copter/plane/rover (NuttX) and the board boots fine.  I try to load the latest copter/plane/rover (Pick previous firmware mission planner AC3.3) and the board boots with a solid red LED. 

Im not trying to load copter/plane/rover any old versions but trying to install antenna tracker and its booting up with a solid red LED. You can still get the NuttX console over USB but the board wont boot.

I get that Chibios is the future but if you try to install antenna tracker your installing a binary that over 3 years old. We need a Chibios version of the antenna tracker. If antenna tracker is no longer been supported then that need to be made clear.

**Version**
AC3.3

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
non
**Hardware type**
Im running a pixhawk lite but seems to be the same issue on the pixhawk mini and maybe others
https://discuss.ardupilot.org/t/pixhawk-mini-as-antenna-tracker-solid-red-light-when-booting/25143

**Logs**
non

FLYBYME, 2019-04-24:
I have no installed http://firmware.ardupilot.org/AntennaTracker/latest/Pixhawk1/antennatracker.apj and it seems to work.

I suspect this issue is a bootloader issue.

proficnc, 2019-04-25:
Ac3.5 was never supported on the cube, I suspect that’s the problem.

closed 2019-04-25 by proficnc

rmackay9, 2019-04-25:
I agree that we need to get an antenna tracker release out.  I'm happy to help @IamPete1 and @peterbarker to do this if they ask me to.

IamPete1, 2019-04-25:
I see no reason we cant got to a beta now, there doesn't seem to be any important related PR's waiting. I think a hand full of people are now running the latest firmware from master without any issues.

peterbarker, 2019-04-25:
On Thu, 25 Apr 2019, Peter Hall wrote:

> I see no reason we cant got to a beta now, there doesn't seem to be any important related PR's waiting. I think a hand full of people are now running the
> latest firmware from master without any issues.

I'll ping Randy tomorrow and we'll get a beta out.

FLYBYME, 2019-04-25:
@proficnc Never said it was a cube.

This issue is a bit deeper then just getting the antenna tracker release out. This issue is that you cant install older version of ardupilot if you have installed anything with Chibios.