ArduPilot/ardupilot #11188: Pixhawk v5 SPI6_RX DMA have a question

DZUAV, 2019-04-25:
Pixhawk v5 SPI6_RX DMA have a question

bys1123, 2019-04-25:
Can you describe the details?

DZUAV, 2019-04-25:
I I'm going to open SPI6, but I've got a compilation error

lvale, 2019-04-25:
Support issues at the forum 

discuss.ardupilot.org

thanks

closing

closed 2019-04-25 by lvale

DZUAV, 2019-04-26:
@lvale hi, Please  how to solve this problem.

proficnc, 2019-04-26:
To solve the problem, ask in the forum as @luis said.

DZUAV, 2019-04-26:
@proficnc Do you have his contact information?

proficnc, 2019-04-26:
The forum is discuss.ardupilot.org 

This is where you should ask for any assistance.

lvale, 2019-04-26:
> discuss.ardupilot.org

https://discuss.ardupilot.org

DZUAV, 2019-06-11:
@luis hi

DZUAV, 2019-06-11:
@luis can you help me

lvale, 2019-06-11:
@DZUAV 
Someone will be able to help you if you make a concrete question. Please use the forum at https://discuss.ardupilot.org