ArduPilot/ardupilot #11192: Copter: LOITER_TURNS mission command moves to random position on circle radius

rmackay9, 2019-04-25:
To recreate the issue in SITL do this:

- create a mission like shown below with a LOITER_TURNS command with no lat/lon specified
- arm the vehicle in Loiter ("arm throttle")
- switch to Auto and raise throttle ("AUTO", "rc 3 1100")
- the vehicle will move to some point on the circle
- stop and restart the simulation and repeat the steps above
- the vehicle will move to a different point on the circle

[image]

rmackay9, 2019-04-25:
I think the issue starts in the [mode_auto.cpp's do_circle method](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_auto.cpp#L1199).  the circle_center is being set to the vehicle's current location which is fine but then within [AC_Circle library's init function](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_WPNav/AC_Circle.cpp#L52) (there are two) we should be calling init_start_angle(true);

brotherlymite, 2019-12-24:
Sorry but I'm still a bit confused on what exactly the problem is....
Is it something like when it starts to go to circle around its home position (As not lat/log is provided in LOITER_TURNS) it starts from a different position (or angle) on every try and we want it to start or move from a specific angle or something?

1st RUN
[image]

2nd RUN
[image]

In the second run it started and ended from a different angle than the previous one

brotherlymite, 2019-12-25:
@rmackay9 Is what I'm thinking correct so I can try to fix this issue?

YuxinPan, 2024-08-01:
No longer able to reproduce as of ArduCopter 4.6-dev (see the 2 attempts in the attached video, in both cases the vehicle did not deviate).

I believe I have followed the same mission set up and SITL commands in my attempt.

Perhaps this issue has been unknowingly fixed in the last 5 years? @peterbarker what do you think?

https://github.com/user-attachments/assets/02199de9-b11c-4c54-a443-203c6f24a257

https://github.com/user-attachments/assets/876ef901-2a4f-45b0-a68b-c48bd25f5651

peterbarker, 2024-08-01:
Thanks @YuxinPan - from Discord:
[image]

Closing this now - many thanks for re-testing!

closed 2024-08-01 by peterbarker