ArduPilot/ardupilot #11193: Rover: improve navigation using S-curves

rmackay9, 2019-04-25:
This item is on the [2019 roadmap](http://ardupilot.org/dev/docs/roadmap.html) and [Leonard Hall did a video presentation](https://www.youtube.com/watch?v=LHq5o9zgNWk) about it at the 2019 developer un-conference.

Rover's navigation routines have a few issues:

- the L1 controller's response is based on the vehicle's **current speed** instead of it's **desired speed**.  This means an oscillating throttle/speed can lead to a lateral wobble ([JohnE's issue](https://discuss.ardupilot.org/t/the-greatest-puzzle-of-all-time-path-wobble/36316)).
- the chain of execution is throttle/speed controller -> lateral controller (L1 controller) -> steering controller.  A better setup would be to have a waypoint navigation library (aka path planner) that calculates both desired speed and turn-rate and then forwards these to the throttle/speed and steering controllers (the order shouldn't/wouldn't matter).
- corners are not planned.  Instead a variety of parameters (TURN_MAX_G, TURN_RADIUS, WP_OVERSHOOT, WP_SPEED, NAVL1_ etc) can be tuned so the speed and cornering works.  A better setup would be for the waypoint navigation library to decide a path and then the resulting desired speed and turn rates are executed by lower level controllers.

A first step to implement all this is to create an AR_WPNav library similar to Copter's AC_WPNav library (but much simpler).  This should have methods including get/set_destination(), set_default_speed_xy(), update(), reached_destination and methods to retrieve the results including get_desired_speed() and get_desired_turn_rate().  Some methods from mode.h including calc_steering_to_waypoint() and calc_steering_to_heading() should be replaced with calls to this new AR_WPNav library.  Parameters including WP_RADIUS, WP_SPEED, WP_OVERSHOOT and (maybe) TURN_MAX_G should be moved into this new library.

This [wiki page explains Rover's L1 controller](http://ardupilot.org/dev/docs/rover-L1.html).
This change will probably resolve this related issue: https://github.com/ArduPilot/ardupilot/issues/7386

Linjieqiang, 2021-02-28:
What's the progress now?

rmackay9, 2021-03-02:
@Linjieqiang, The 2nd and 4th item have been done but the 1st and 3rd have not.  The big improvement will come with the introduction of S-Curves but that will only happen after it has been completed for Copter (see PR https://github.com/ArduPilot/ardupilot/pull/15896)

IamPete1, 2022-09-11:
@rmackay9 Any update?

rmackay9, 2022-09-11:
@IamPete1, yes this is all done and included in 4.3.  txs!

closed 2022-09-11 by rmackay9