ArduPilot/ardupilot #11210: DO_SET_SERVO ignored if channel set to RC_PASSTHRU

sgofferj, 2019-04-27:
### Bug report

**Issue details**
Not sure if it's a bug or a feature but I open it as a bug. Label can be changed if necessary.
I have a big copter with various additional functions, such as lights, servo tilts, cargo locks, etc. The additional functions are connected to the fc's servo outputs and those are configured as RC_PASSTHRU so I can control everything from the tx. However, I cannot control those functions in a mission. Any DO_SET_SERVO commands seem to be ignored.
If this is the intended behavior, I'd like to suggest a change to the effect that as soon as a DO_SET_SERVO command is issued in a mission or manually through the GCS software, the command overrides any input from the tx and tx control of that channel is disabled until disarm or a cancellation command, e.g. DO_SET_SERVO <servo>,-1. That would in essence replicate the DO_SET_ROI behavior for the gimbal.

**Version**
Copter 3.6.7, Copter 3.6.8

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4

**Hardware type**
Pixhawk

**Logs**
No relevant logs available at the moment but can be produced if necessary.

magicrub, 2019-04-29:
Interesting idea, thanks for the request!

sgofferj, 2019-04-30:
Oh, my second "good first issue" :D :D :D

magicrub, 2019-04-30:
"first issue" meaning someone who wants to work on the code but doesn't know where to start. This issue seems to be a good one to get started on without having to be an expert on the whole codebase

sgofferj, 2019-05-01:
AH! Like a good first project for a new developer! Got it!

OXINARF, 2019-06-13:
Done by #11569 (which superseded #11468). Thanks to @tatsuy and @rmackay9!

closed 2019-06-13 by OXINARF