ArduPilot/ardupilot #11228: copter: NTF_DISPLAY_TYPE 1 is affecting FrSky SPort Passthrough on Serial 1

zenuavsolutions, 2019-04-30:
### Bug report

when setting NTF_DISPLAY_TYPE to 1, FrSky SPort Passthrough (Serial1_PROTOCOL = 10) gets no more data. 

_Please describe the problem_

**Version**
Copter 3.7.0 dev

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Cube Black

**Logs**
-

zenuavsolutions, 2019-05-07:
it has something to do with connecting the oled display to i2c on GPS1 port, when using i2c on GPS2 s-port passthrough is working again
also when NTF_DISPLAY_TYPE is 1 and the display is not connected passthrough is working
...curious

WickedShell, 2019-06-27:
This is almost certainly due to the problems described in #6274

closed 2019-06-27 by WickedShell