ArduPilot/ardupilot #11249: ArduPlane: startup importat console messages missed

Jaaaky, 2019-05-02:
### Bug report

**Issue details**

_I've found that we miss initial console messages displayed at this line https://github.com/ArduPilot/ardupilot/blob/48aacb94629a07abae630c3dd0c9ad57cca13731/ArduPlane/system.cpp#L26 and after.
Apart from missing vehicle name, firmware info and free RAM which may not be critical; missing messages from the following load_parameters() is problematic, specially if params are got erased and don't know why for example.
Also some subsequent importatnt messages may got missed.
Debugging the problem; I've found that adding hal.scheduler->delay(1000); after init_console() solved the problem. Smaller delay values didn't work. But it's a bad idea to lose a whole second here which can affect watchdog resets too.
 I think it's related to a buffer flush issue; I don't know how to solve.
But I found some hack that did fix it for me; printf ~220to255 newlines to the console; it's too fast and works around the problem. printf \r or NULL and different other characters didn't had the same effect.

Here is my patch if it could be useful - also added some nice banner there_

**Version**
_master and many other older branches_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk

Tested with `mavproxy --nowait`

Jaaaky, 2019-05-02:
Yes. Usual, no?
It's default, nothing special, I think.

OXINARF, 2019-05-02:
Yes, usual, but that's the reason why you don't see the messages. The messages aren't lost, they are sent before USB has a chance to be connected. If you change the console to be on a UART you would see the messages.

Jaaaky, 2019-05-02:
But why would sending that series of new line char activates it that fast? Within 1900 microsecond?

magicrub, 2019-05-03:
those messages that early can only be seen if you OS is holding the port. Ubuntu does that just fine, Windows does now. Not sure about MacOS. What OS are you using?

Jaaaky, 2019-05-03:
Solus Linux, not Ubuntu. This occurs even on reboot command while mavproxy is holding the port.
Can you, please, confirm that you can see Firmware version and load_all messages on Ubuntu?

OXINARF, 2019-05-03:
@Jaaaky it's introducing a delay, apparently 1900us is enough - the USB port is setup way earlier than that so at that point in code I'm not surprised that's enough time to have it connected to the computer.

Jaaaky, 2019-05-03:
> @Jaaaky it's introducing a delay, apparently 1900us is enough - the USB port is setup way earlier than that so at that point in code I'm not surprised that's enough time to have it connected to the computer.

Yeah, But if you just set schedular->delay() instead, you've to wait to ~1000ms to get the same result. I've tried values up to 600ms with no success.

peterbarker, 2020-11-25:
These messages are now all send-texts - so at least we'll have different bugs if this is still a problem.

closed 2020-11-25 by peterbarker