ArduPilot/ardupilot #11271: Matek f405-STD reboot in flight

kiryam, 2019-05-04:
### Bug report

**Issue details**

I had crash today, it looks like reboot in flight

**Version**
3.10.0-dev (b1bc4268)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
airplane

**Hardware type**
MATEK f405-STD

**Logs**
[00000001.BIN.txt](https://github.com/ArduPilot/ardupilot/files/3144841/00000001.BIN.txt)

https://youtu.be/2uI3km0nMs4

kiryam, 2019-05-05:
I found that I have stalled sero on left wing. But it stalled in neutral position. Look like servo was stalled. Controller reboots and moved servo on the right wing up. after that I had a crush.

closed 2019-05-10 by kiryam

fikusir, 2020-06-03:
Just happenedto me as well. Have you find the reason why it happened to you? Me not

rmackay9, 2020-06-04:
@fikusir, 

reboots can happen for a few reasons - the most common is probably a power issue but recent versions of ArduPilot include an "[independent watchdog](https://ardupilot.org/plane/docs/common-watchdog.html)" that records a message in **the next log** after the flight that ended in a reboot.

If you had a telemetry connection you may also have seen "Watchdog" appear on the HUD or OSD.

So ideally it would be good to see the logs of the flight in question and the next log if possible.

ValeraSokot, 2023-06-24:
Привет. У меня точно такая же проблема. Получилось ли тебе найти причину перезагрузки ?

ValeraSokot, 2023-06-24:
https://www.youtube.com/watch?v=SIjuAKOhuOg&ab_channel=%D0%B2%D0%B0%D0%BB%D0%B5%D1%80%D0%B0%D1%88%D0%BE%D0%BA%D0%BE%D1%82

kiryam, 2023-06-24:
@ValeraSokot I think problem was been with servo. It locks on max position -> current draws goes high -> power module overloaded.

ValeraSokot, 2023-06-24:
так ты заменил серву и проблема ушла ?

kiryam, 2023-06-24:
Sorry, I don't remember already.

ValeraSokot, 2023-06-24:
Плохо конечно что не помниш , ладно спасибо буду искать дальше.