ArduPilot/ardupilot #11274: hwdef_bl: Use different board_id for different boards in ChibiOS Bootloader

bys1123, 2019-05-05:
### Feature request

**Is your feature request related to a problem? Please describe.**
Current state lots of boards share the same board_id, but it maybe cause them can't work well. Especially for new users.
[image]
[image]

**Describe the solution you'd like**
I think we should use different board id for different board that already listed in AP_HAL_ChibiOS, then GCSs might possible identify it. If not, may cause a lot of unexpected behaviors.

**Platform**
[√] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

bys1123, 2019-05-05:
cc'ing @DonLakeFlyer

tridge, 2019-05-13:
Yes, a different board_id for each board would have been a good idea, but if we changed the board_id now then existing boards would not be able to accept the new firmware.
The fix we have in the ArduPilot bootloader is to encode the board name (like "Pixhawk4") in the bootloader string. That is a much better solution I think, as it is backwards compatible

bys1123, 2019-05-13:
@tridge Sounds good! Thanks!

closed 2019-05-13 by bys1123