ArduPilot/ardupilot #11278: AP_OpticalFlow: only publish PX4FLOW data to telemetry on start then stopped

bys1123, 2019-05-06:
#### Issue details

MP can only show the first frame from optical flow. But the log shows it works well.

#### Version
Copter-3.6.7~3.6.9

#### Platform
[  ] All
[  ] AntennaTracker
[√] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
Pix32(Pixhawk), Pixhawk4

#### Logs

[image]

tlog:
https://www.dropbox.com/s/8mos4rk87wtbm38/2019-05-04%2021-15-39.tlog?dl=0
https://github.com/ArduPilot/MissionPlanner/issues/2141#issuecomment-489495783

other person have the same issue:
https://discuss.ardupilot.org/t/how-to-setup-and-check-optical-flow/20444/40

bys1123, 2019-05-12:
[image]
[image]
Tried CX-OF today, values of opt_* still not updating.

bys1123, 2019-05-22:
This drone also can't get rangefinder's data.
But I tried on other drone, use same firmware and same hardware configuration, that works fine.
Is there a parameter control this?

bys1123, 2019-05-22:
I think I found the reason, disconnect companion computer, then optical flow information will work.

bys1123, 2019-05-22:
It's work now, what ever I connect companion or not.

bys1123, 2019-05-22:
But ESC telemetry can't show the data now……

bys1123, 2019-05-22:
The reason why it's not have optical flow status data reading because of I didn't connect rangefinder, I got output after I connect a VL53L1X lidar, Is this expected? @rmackay9

rmackay9, 2020-01-10:
I think we've resolved this in Copter-4.0.0-rc1 (https://github.com/ArduPilot/ardupilot/pull/13182) so I'm going to close this if that's OK.  Happy to re-open if we find the issue persists.

closed 2020-01-10 by rmackay9