ArduPilot/ardupilot #11284: Rover: Guided mode's wp_bearing is incorrect for some sub modes

rmackay9, 2019-05-07:
While working on this PR: https://github.com/ArduPilot/ardupilot/pull/11283 I found that Guided mode's wp_bearing valueis incorrect for some submodes.

Guided uses the [mode.cpp's wp_bearing method](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode.cpp#L163) which calls the navigation library. This works when _guided_mode == Guided_WP but is incorrect when it's Guided_HeadingAndSpeed, Guided_TurnRateAndSpeed and Guided_Loiter.

We should override the wp_bearing for Guided mode and then handle these other sub modes.  For example, for Guided_HeadingAndSpeed we should use the target heading.  For Guided_TurnRateAndSpeed we should return 0.0f.  For Guided_Loiter we should call ModeLoiter's wp_bearing.

rmackay9, 2019-05-09:
Also Guided mode continues to report the distance-to-destination even though the rover has stopped.  Look for stop_vehicle() in mode_guided.cpp when the _guided_mode == Guided_WP.

rmackay9, 2019-05-09:
Similarly @peterbarker commented during the review of this PR:https://github.com/ArduPilot/ardupilot/pull/11283 that Mode::get_desired_lat_accel() is not always correct for Guided submodes in cases where the AR_WPNav library is not being used.  Comments are below.

> This is probably not correct in that guided mode marks itself as an autopilot mode when it isn't necessarily using the waypoint navigation controller leading to essentially undefined values being returned here. It may still have a desired acceleration, just not stored in wp_nav.

aliakbarkhan, 2022-03-06:
> While working on this PR: #11283 I found that Guided mode's wp_bearing valueis incorrect for some submodes.
>
> Guided uses the [mode.cpp's wp_bearing [...]

is this issue still open i would like to work on it

peterbarker, 2026-11-15:
Thanks to all that contributed to fixes for this issue.

Closed by [PR #19220](https://github.com/ArduPilot/ardupilot/pull/19220) and [PR #34493](https://github.com/ArduPilot/ardupilot/pull/34493). Rover Guided now overrides wp_bearing, nav_bearing, crosstrack_error and get_desired_lat_accel per submode (74889f1c7a, [PR #19220](https://github.com/ArduPilot/ardupilot/pull/19220)): Loiter delegates to ModeLoiter and the non-wp_nav submodes return 0 instead of stale values. HeadingAndSpeed reports its target in nav_bearing (ac1fd4b03a, 9b80636098, [PR #34493](https://github.com/ArduPilot/ardupilot/pull/34493)). Closing; please reopen if a submode still reports wrong values.

closed 2026-11-15 by peterbarker