ArduPilot/ardupilot #11286: Rover: In Guided mode's HeadingAndSpeed mode the vehicle should slow for large heading changes

rmackay9, 2019-05-07:
In Guided mode's HeadingAndSpeed mode, the vehicle will mostly independently try and turn towards a specified heading and attain the specified speed.  Ideally if the heading angle change is very large, the vehicle should reduce it's speed so that it does not do a very wide turn.

The calculation could be similar to the "lane based speed control" used for waypoint navigation (see AR_WPNav).   We should use the WP_OVERSHOOT but we shouldn't try to use an origin and destination.

mikerob, 2019-05-07:
It is worth parameterising that as an option - to either maintain speed (and turn as wide as required) or maintain navigation and slow as required?

rmackay9, 2019-05-07:
@mikerob, perhaps if someone says they want both behaviours.. my guess though is that people would be happy with the vehicle slowing to a speed so that it turns within a radius of the WP_OVERSHOOT parameter value.

By the way, I suspect we will move the HeadingAndSpeed control out of the Mode class into the new AR_WPNav class as a follow up to [this PR](https://github.com/ArduPilot/ardupilot/pull/11283).