ArduPilot/ardupilot #11294: Copter: Master logging fails consistently

lvale, 2019-05-07:
With both a CUAV v5 and a mRov2.1-777 so both F7 boards, after a short time the logger starts failing (on Pre-Arm which would stop flying)

Just have a couple of previous large log files on the sd card, and the board after a while starts sending logging failures (note: both boards on the table with nothing connected but the usb cable)
[image]

example parameter file:

[mav.parm.txt](https://github.com/ArduPilot/ardupilot/files/3152817/mav.parm.txt)

@peterbarker @tridge This is the issue we've discussed.

I can replicate this easily every time - Using Master Copter-Heli

Jaaaky, 2019-05-11:
Are using a high quality SD card?
If not, I recommend that you use high quality UHS-I SD card or better, then try again.

Jaaaky, 2019-05-11:
If you're using proper SD card as recommended, can you try to adjust BRD_SD_SLOWDOWN and try again?
Try increasing the value up to 32.

lvale, 2019-05-11:
@Jaaaky Thanks for the input, but those are the obvious points, but not on this situation.

IamPete1, 2021-08-29:
@lvale any update?

rmackay9, 2021-08-29:
We've done a lot of fixes to the logging system since this report.

lvale, 2021-08-29:
Would need to try to replicate with next release

IamPete1, 2021-08-29:
@lvale If you could test on 4.1 beta that would be great.

lvale, 2021-08-29:
Don't promise

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1