ArduPilot/ardupilot #11296: Watchdog reset copter in flight without actual problem

Pedals2Paddles, 2019-05-08:
### Bug report
Copter master flying FMUv3 (CubeSolo build target) on a stock Solo Pixhawk 2 cube.  Watchdog reset the FC while the copter was in flight for seemingly no reason causing it to crash. The copter was under control, flying normally, and there are no errors or irregularities in the log that I can readily see.  The operator was flying normally, the motors stopped, and while falling the reboot tones were audible.  The log file that began after the reboot contains `2019-05-06 23:28:17 PreArm: Internal errors detected (0x800)` which indicates the FC rebooted by watchdog.

**Version**
Copter 3.7-dev from 4/28/2019 (FMUv3)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
3DR Solo quadcopter

**Hardware type**
Pixhawk 2.0 Cube

**Logs**
[watchdog reset.zip](https://github.com/ArduPilot/ardupilot/files/3155023/watchdog.reset.zip)

AndKe, 2019-05-08:
Here are the logs from the controller: https://drive.google.com/open?id=1xUPXcHYreoGjBLhYHnRCnypsT4LNE1hu

robustini, 2019-05-09:
I think IWDG should be completely removed from the Copter because in any case it might help.
With Plane certainly has reason to exist, but on the Copters it would not produce any useful result.
Alternatively it could be left enabled only when the system is unarmed.

rmackay9, 2019-05-09:
Actually the watchdog stuff could provide some real benefits to Copter as well.  In particular it allows us to determine if an in-air failure was the cause of a software bug or a brownout.  Just knowning this difference would be very valuable.  I agree that for Copters we aren't ready to support re-arming in flight because there is too little time and we haven't successfully tested this works yet.

robustini, 2019-05-09:
I'm aware that this can absolutely be useful for debugging, that copter would have crashed even without IWDG, the important thing is to educate users and inform them of this.

peterbarker, 2019-05-09:
On Wed, 8 May 2019, Marco Robustini wrote:

> I think IWDG should be completely removed from the Copter because in any case it might help.
> With Plane certainly has reason to exist, but on the Copters it would not produce any useful result.

Depending on the vehicle, quiescing the motors before impact may make the 
vehicle significantly less dangerous.

> Alternatively it could be left enabled only when the system is unarmed.

The IWDG can not be disabled once it is enabled.

Pedals2Paddles, 2019-05-09:
In this instance, I have no reason to believe the copter would have crashed even without the watchdog.  The copter was flying just fine and under normal responsive operator control.  I'm not aware of any past instances of ArduCopter suffering an FMU or IOMCU lockup when installed on quality autopilot hardware like a real pixhawk or cube. And yet, mere days after installing the new watchdog for the first time, it experiences a lockup and watchdog caught it?  That would be an improbable coincidence.

From everything I can see, the watchdog triggered unnecessarily and caused the crash.  If a safety feature is causing more crashes than it prevents, that's a big problem.  And so far it is saves 0, crashes 1 :).

Until the watchdog is a little more robust (or a more concrete cause is determined), I've set parameter `BRD_OPTIONS` to 0 to disable the watchdog on the solo.

tridge, 2019-05-10:
@peterbarker just noticed that the 2nd baro didn't appear on the reset boot. That is very interesting. We'd like you to try booting this board several times and see if the 2nd baro is gone forever, or if it only disappeared this one time.
I'd also like to get the tlog as well if we can, as that may give us a clue as to why the 2nd baro didn't get found.
The baro not appearing may indicate something went wrong in one of the following areas:
 - SPI transfer on SPI1
 - MS5611 baro driver
 - hardware fault on SPI1

Pedals2Paddles, 2019-05-10:
@AndKe I sent you a few messages on gitter.  If you can get the tlogs off the controller, and see Tridge's comments above regarding the baro.

AndKe, 2019-05-10:
@tridge @Pedal2Paddles   , Sorry for the late reply, I am in China for UAV business, and slow to communicate due to the great firewall of China and VPN throttling they do.
Solex was used, tlog not saved, but I did upload the controller logs in the second post to this thread, hopefully you can use them, even thay are not in .tlog format

Pedals2Paddles, 2019-05-13:
The T-Logs from the incident are in this zip.  Not sure which of the three has the accident flight, I'm not in a position to open it right now.
[T-Logs.zip](https://github.com/ArduPilot/ardupilot/files/3175129/T-Logs.zip)

AndKe, 2019-05-15:
This is a indoor test flight using the same Solo that fell twice outdoor (the second crash had watchdog disabled, so there is even less data, it rebooted on liftoff, no damage)
https://drive.google.com/file/d/1IIm6uTbzLWo41WQ-WfvV46tdqZw-zpsX/view?usp=sharing
The log is collected hovering 1-10cm over the concrete/rebar floor, with GPS repeater, so not perfect.
Later on I climbed a >4m by holding solo in the leg, walking up a stair to mezzanine just to provoke in-flight yaw realignment.  
I flew two flights like that (recorded tlog of this one using APMPlanner2)
I do not know why there is so much statustext , ( Solex was simultaneously connected)

So it fell out of the air twice within 2 minutes outdoors, and I flew many times that long indoors without issue.

Pedals2Paddles, 2019-05-15:
Wait, so you had another crash where it rebooted in flight with `BRD_OPTIONS` set for 0???

Tridge's enhanced watchdog logging has been merged into master.  If you want to grab the latest CubeSolo build off the ArduPilot firmware server and fly it, it will provide lots more detail the next time it has a failure.

AndKe, 2019-05-15:
While I was away, one of my colleagues tried that parameter for me few days ago.
He did barely arm, not getting of the ground as it rebooted. the log just ended, no harm done.
I flew it in the lab today, a lot, first with no changes, then corrected gimbal settings and such, no issue.
I am about to take it outside again.   - maybe it will just crash outside :)

If I should try current master, it would be nice if you pointed me to a prebuilt binary, so that nothing in my build environment cannot be an issue.

Pedals2Paddles, 2019-05-15:
Latest copter master CubeSolo build APJ file can be found here: http://firmware.ardupilot.org/Copter/latest/CubeSolo/

tridge, 2019-05-16:
@Pedals2Paddles now we know this issue definitely happens with watchdog disabled, perhaps we can enable it again?

Pedals2Paddles, 2019-05-16:
Yep.  Especially given the new logging you've added, and your confidence that it wasn't a false positive.  I'll be updating the parameters for the build target this weekend to re-enable it.

AndKe, 2019-05-16:
I flew the Solo#7 again to see if I could make it happen.  (yes it did)
After long hovering above bushes, I flew a few hundred meters away over some moss/wet areas.
Suddenly the sound stopped, I could still see the video stream of falling Solo.
The Solo crashed - when I walked up to it, one of the arm LED's was flashing, and one was off, otherwise normal colors.
The controller was "Waiting for Solo" - not able to speak with it.
The gimbal was limp.
I did shutdown the Solo - battery cut power, but there were no "shutdown sound".
Restarted the Solo, it connected to controller just fine, gimbal was working again.
I could arm and ran it for a while with no propellers, (/hence the last log)
All material in:
https://drive.google.com/drive/folders/1p5WtjGcaPtyJnaElNbE-bIUnjUQoKX43?usp=sharing
please note that there are logs from controller, Solo, and Solex.  - along with onboard Gopro and crash site photos.

AndKe, 2019-05-16:
After the crash in RECENT.BIN, please note SubSys25 ECode 1 and "Potencial Thrust Loss" 
recent kern.log files have more  
`3dr_solo user.err kernel: imx-uart 21e8000.serial: Rx FIFO overrun`  lines

Pedals2Paddles, 2019-05-16:
@tridge @peterbarker  These are the dataflash logs for the most recent crash and after crash.
[DF Logs.zip](https://github.com/ArduPilot/ardupilot/files/3189387/DF.Logs.zip)

In the video, the motors just stop in flight while hovering.

AndKe, 2019-05-28:
-I were unable to reproduce crash on another Solo with same config.
-Two days ago I took some time to finely search for issue with that Solo.  
I found poorly seated mezzanine connector on the FPC cable to the damped part of the Solo cube.
Reseated it, flew aggressively three batteries today, no issue.
The crash must have been due to loss of multiple sensors.

Pedals2Paddles, 2019-05-28:
Well that would definitely do it!  With that, I will close this as solved!

closed 2019-05-28 by Pedals2Paddles

peterbarker, 2019-05-28:
Thanks @AndKe - appreciate you digging further.

This does mean we have a problem, however.  Hopefully we can reproduce it on a different cube - but if we aren't, would you be happy to fiddle with yours if we give you binaries with more logging?

AndKe, 2019-05-29:
@peterbarker Yes. I have more cubes and can try later on reproducing this issue.
Right now, I do not know whatever AP could fly with the loss of the mezzanine PCB that would depend on what sensors are on the main board, and where the  I2C pullups are located.
Regardless , this makes me think of how AP will handle losing sensors , be it external sensors due to connection loss, or even internal sensors due to a I2C short or malfunction holding SDA down.
Do we even know what happens if external devices like magnetometers, LEDs, pitot, lidar, suddenly stop responding ? - will something wait too long at let watchdog expire?

I would absolutely love to see, and test ,  future development that aim to mitigate consequences of sensor loss.

AndKe, 2019-06-19:
Another crash, this issue should maybe be reopened  Solo#7 went down the same way as #4 did before.
Same firmware/software.

I checked the suspected mezzanine connector, and it does not seem to be the reason. it is hard to inspect on a closed cube, but if the cube is assembled without the black casing, one can see that the connector is actually kind of secured in place due to the sandwiched PCB's

All logs here: https://drive.google.com/file/d/1PvRKcxEqw4ihlFE15Hr36bZKx5h4H3Nw/view?usp=sharing

nothing spectacular... it just stopped motors(and avionics) for no apparent reason.

Pedals2Paddles, 2019-06-20:
@tridge @peterbarker Here is the data flash and tlog from the above.  This not the current master. This is the same FW as was used in the first incident.

[Solo Shudown.zip](https://github.com/ArduPilot/ardupilot/files/3308353/Solo.Shudown.zip)