ArduPilot/ardupilot #11299: Copter:Should WPNAV implement the coordinating steering in AutoMode?

thomasonegd, 2019-05-08:
### Feature request

if the copter's WPNAV add coordinating steering support,I think it'll becomes more smoother & more efficient during the mission.
[image]

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

auturgy, 2019-05-08:
Is what you’re asking for different from Spline Waypoints?
http://ardupilot.org/copter/docs/mission-command-list.html

thomasonegd, 2019-05-08:
not like the spline waypoint, the spline waypoint will accros outside the border of the wps

bnsgeyer, 2019-05-08:
@thomasonegd Are you concerned with the fact that it slows down before entering the turn and then yaws and finally speeds up coming out of the turn?  Is your desire to have it maintain speed into the turn and effectively fly the turn like an airplane?  
I am one of the developers on the traditional helicopter code.  From a helicopter standpoint, missions are flown at much higher speeds (~20-30 m/s).   Regular waypoints can't be used for surveying these speeds as the helicopter will slow down and then speed up.  Survey planning with spline waypoints are hard to do in order to keep the aircraft at speed.  My Co-maintainer has developed techniques to use spline waypoints for surveys at high speed but it is very tedious.  
Recently we have been working to integrate the plane L1 navigation controller but only make it available to helicopters.  there has been some pushback on integrating this kind of navigation controller in copter.  Leonard is looking at integrating a trigonometric S-turn navigation.  This will round the corners more on turns but will still cause the aircraft to slow down coming into the turn.  So it will probably do what you are saying but with some change in airspeed.  Here is a link to Leonard's [presentation.](https://www.youtube.com/watch?v=LHq5o9zgNWk)

thomasonegd, 2019-05-09:
@bnsgeyer yeah,my opnion is the same with yours,like the " it slows down before entering the turn and then yaws and finally speeds up coming out of the turn".I have seen Leonard's presentation, but I found that it is only metion on ROVER recently. I wonder if there is any plan to merge it into the Copter.If so, when will it be.

rmackay9, 2019-10-26:
I think this is yet another variation of this issue: https://github.com/ArduPilot/ardupilot/issues/6194

Thanks for the report.  I think I'll close this because all roads lead back to @lthall 's S-curves.

closed 2019-10-26 by rmackay9