ArduPilot/ardupilot #11320: ArduPlane SITL: JSBSim not properly shutdown when terminating simulation

simfinite, 2019-05-10:
### Bug report

When running SITL with `-f JSBSim`, then exiting the simulation with `CTRL-C`, the JSBSim process is not terminated. This process blocks UDP ports and any subsequently started simulation will fail to connect with `link 1 down`. A `killall JSBSim` before starting the simulation resolves the issue but we should find a better solution for this.

This problem has originally been reported [here](https://github.com/ArduPilot/ardupilot/issues/10612#issuecomment-486420509) and [here](https://github.com/ArduPilot/ardupilot/pull/11217#issuecomment-490939275). Also, it may be related to issue #10995.

**Version**
latest master / latest ArduPlane release (3.9.8)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n.a.

**Hardware type**
n.a.

**Logs**
n.a.

rajat2004, 2019-05-10:
Maybe the simulation endtime can be set using `--end`
https://github.com/JSBSim-Team/jsbsim/blob/master/src/JSBSim.cpp#L772

When we get the Ctrl+C command, before shutting down the SITL, an end-time can be sent which would be just after the current simulation time and then let JSBSim handle closing itself

The setting of the time just after the current time is possible since we know the current simualtion time  (The PRs #12117 - specifically commit https://github.com/ArduPilot/ardupilot/pull/11217/commits/552644845402bd42151eb028a6d66226a7853d36 and the corrsponding JSBSim PR have been merged) 

Nice catch on #10995, I had noticed that earlier but didn't connect to this issue

rajat2004, 2019-05-10:
Sorry, wrong suggestion. I should stop doing these things when sleepy.
The earlier suggestion of using `end` only works in the beginning, i.e it can only be set in the beginning. 

Instead `quit` command can be sent just like how `iterate` is sent.
https://github.com/JSBSim-Team/jsbsim/blob/master/src/input_output/FGInputSocket.cpp#L230

simfinite, 2019-05-10:
> Instead quit command can be sent just like how iterate is sent.
Sounds good. But how to trigger on exit? From an OO point-of-view the destructor would make sense. 

As a first guess, I tried to add a virtual d/tor to `class Aircraft` and have the member `SITL::Aircraft *sitl_model;` deleted in the d/tor of `class SITL_State` which in turn is initialized as a static variable in `HAL_SITL_Class.cpp` ... at least that's my current understanding. However, no d/tor is being invoked at all, so I must be missing something.

I wonder, how this worked with the old JSBSim? Did it just exit by itself as soon as it didn't receive any updates? I don't remember ever seeing any exit commands on ArduPlane side.

rajat2004, 2019-05-10:
I think I've identified the problem.
I went looking into `sim_vehicle.py` to see how it handles exiting. Turns out it does exactly what we do, kills all the processes. 
https://github.com/ArduPilot/ardupilot/blob/2c519d5176a9c2ea1990ecdb234c633be5e563b6/Tools/autotest/sim_vehicle.py#L217

It calls the function `kill_tasks_psutil()` which kills all the processes with the name in the victim list. But when I added a line to print the process names which it's killing, nothing is happening!
The problem is `proc.name` doesn't just give the name, it gives a whole lot of other info as well. Instead, we need to use `proc.name()` to just get the string of the process name. I tried it and it works! JSBSim gets killed when we close the SITL.
https://stackoverflow.com/questions/23991510/psutil-getting-process-name-is-blank

The above change happened after sim_vehicle.py was written so that might have caused the problem. The doubt which I have is how hasn't this been noticed till now & if this was the problem, then how did the earlier version of JSBSim quit itself?

rajat2004, 2019-05-10:
> The above change happened after sim_vehicle.py was written so that might have caused the problem. The doubt which I have is how hasn't this been noticed till now & if this was the problem, then how did the earlier version of JSBSim quit itself?

This is wrong, the above mentioned API change happened in 2014 while in the file, it's written November 2016 so the the doubt becomes even more valid

rajat2004, 2019-05-10:
Got it, this was changed in the following commit 
https://github.com/ArduPilot/ardupilot/commit/53df307c2d320c7f0d057359973be5fe4946a680#diff-460241a402d32677bcde6618e6e880de

@peterbarker Is this still applicable or can this commit be reverted now?

peterbarker, 2019-05-11:
@rajat2004 Well, the commit message indicates that older Pythons have issues...  so a full revert sounds like a bad idea.  Even just name() vs name is insane - I'm concerned if we revert that then older Pythons may not work :-(

Would it be possible to filter the output of the name() call to get just what we need?  That may be the most compatible solution.

I'm also wondering if name changed at some stage to give you all of argv rather than just the process name....

Also - are we talking Python2 or Python3?

rajat2004, 2019-05-11:
@peterbarker I can try getting the process status and name from the output.
But the thing is that the change in `psutil` happened in 2014
https://github.com/giampaolo/psutil/blob/master/HISTORY.rst#200

I feel that it's already long enough, and I'm not sure people still use such older versions of python, but maintaining backwards compatibility is also important.

The new behaviour is in both Python 2 & 3

rajat2004, 2019-05-16:
PR #11342 has been merged. I think this issue, as well as the probably connected issue #10995, can be closed

simfinite, 2019-05-19:
Great. Thanks everyone!

closed 2019-05-19 by simfinite