ArduPilot/ardupilot #11325: Rover:status rpm is always 0

xlla, 2019-05-11:
### Bug report

**Issue details**
I have follow Rover instruction to set up my rover,  I was configured the parameters WENCxxx , EKxx, AHRSxxx carefully, when I have arm the rover and got two wheels spinning, the rpm1 and rpm2 always shows 0 in status tab in mission planner! but the GroundSpeed seems right!

Since I was use motor driver TB6612FNG, it's need two relay pin to operate one motor, then I have to modify AP_MotorsUGV.cpp to send four relay signal to two motor, but I don't think that modification cause the issue.
 

**Version**
ArduRover V3.5.0-dev (47d8435c)
Mission Planner 1.3.66

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X ] Rover
[  ] Submarine

**Airframe type**
tank with two brushed motor

**Hardware type**
F4BY,  pixhwak
connected to osx by usb cable, then passthru to vmware fusion guest os windows 10.

**Logs**
I am sorry , I cannot get sd card to properly work now, so I turn off the log totally.  I have tried BRD_SD_SLOWDOWN, but no lucky.

magicrub, 2019-05-11:
please post your params. This is likely just a setup issue.

xlla, 2019-05-12:
Hi @magicrub, thanks for reply, here is my relative params,
`
AHRS_EKF_TYPE,3
EK2_ENABLE,0
EK3_ABIAS_P_NSE,0.003
EK3_ACC_BIAS_LIM,1
EK3_ACC_P_NSE,0.35
EK3_ALT_M_NSE,2
EK3_ALT_SOURCE,0
EK3_BCN_DELAY,50
EK3_BCN_I_GTE,500
EK3_BCN_M_NSE,1
EK3_CHECK_SCALE,100
EK3_EAS_I_GATE,400
EK3_EAS_M_NSE,1.4
EK3_ENABLE,1
EK3_FLOW_DELAY,10
EK3_FLOW_I_GATE,300
EK3_FLOW_M_NSE,0.25
EK3_GBIAS_P_NSE,0.001
EK3_GLITCH_RAD,25
EK3_GPS_CHECK,31
EK3_GPS_TYPE,0
EK3_GYRO_P_NSE,0.015
EK3_HGT_DELAY,60
EK3_HGT_I_GATE,500
EK3_IMU_MASK,1
EK3_LOG_MASK,1
EK3_MAG_CAL,2
EK3_MAG_I_GATE,300
EK3_MAG_M_NSE,0.05
EK3_MAG_MASK,0
EK3_MAGB_P_NSE,0.0001
EK3_MAGE_P_NSE,0.001
EK3_MAX_FLOW,2.5
EK3_NOAID_M_NSE,10
EK3_OGN_HGT_MASK,0
EK3_POS_I_GATE,500
EK3_POSNE_M_NSE,0.5
EK3_RNG_I_GATE,500
EK3_RNG_M_NSE,0.5
EK3_RNG_USE_HGT,-1
EK3_RNG_USE_SPD,2
EK3_TAU_OUTPUT,25
EK3_TERR_GRAD,0.1
EK3_VEL_I_GATE,500
EK3_VELD_M_NSE,0.7
EK3_VELNE_M_NSE,0.5
EK3_VIS_VERR_MAX,0.9
EK3_VIS_VERR_MIN,0.1
EK3_WENC_VERR,0.1
EK3_WIND_P_NSE,0.1
EK3_WIND_PSCALE,0.5
EK3_YAW_I_GATE,300
EK3_YAW_M_NSE,0.5
WENC_CPR,32767
WENC_PINA,56
WENC_PINB,57
WENC_POS_X,0
WENC_POS_Y,-0.065
WENC_POS_Z,0.06
WENC_RADIUS,0.033
WENC_TYPE,1
WENC2_CPR,32767
WENC2_PINA,58
WENC2_PINB,59
WENC2_POS_X,0
WENC2_POS_Y,0.065
WENC2_POS_Z,0.06
WENC2_RADIUS,0.033
WENC2_TYPE,1
RELAY_DEFAULT,0
RELAY_PIN,52
RELAY_PIN2,53
RELAY_PIN3,54
RELAY_PIN4,55
RELAY_PIN5,60
RELAY_PIN6,-1
BRD_SD_SLOWDOWN,0

`

rmackay9, 2020-12-11:
I think I'll close this because Rover-4.0 (and higher) reports the wheel encoder information using the WHEEL_DISTANCE message as documented on the wiki. https://ardupilot.org/rover/docs/wheel-encoder.html.

I'm sure these new messages are working and we are very unlikely to go back and fix an older release.

closed 2020-12-11 by rmackay9