ArduPilot/ardupilot #11327: Error during compilation using waf (Compiler error)

MastaMinds, 2019-05-11:
#### Issue details
Hello Ardupilot community!

I am trying to build the ArduCopter firmware, but I am having some issues. At first, I was following a Youtube video, but I then realized that things have changed a bit, and I followed up the steps listed on the website, this link:
http://ardupilot.org/dev/docs/building-setup-linux.html 
(I am using Ubuntu 18.10).

Most of the steps were successful, however, I am facing some issues. At first, I didn't know which board to select.. Pixhawk1 or pxf? (I am assuming pxf was px4-v2 previously).
I decided to configure waf for Pixhawk1 (I am using the original Pixhawk with px4 installed). 
But when I first tried to build the firmware I got this error:

> [  9/552] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Sub.cpp
[ 10/552] Compiling libraries/AC_AttitudeControl/ControlMonitor.cpp
[ 11/552] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.cpp
[ 12/552] Compiling libraries/AC_AttitudeControl/AC_PosControl_Sub.cpp
In file included from ../../libraries/AP_Param/AP_Param.h:25:0,
                 from ../../libraries/AC_AttitudeControl/AC_AttitudeControl.h:7,
                 from ../../libraries/AC_AttitudeControl/ControlMonitor.cpp:1:
../../libraries/AP_Common/missing/cmath:3:22: fatal error: cmath: No such file or directory
 #include_next <cmath>
                      ^
compilation terminated.

In file included from ../../libraries/AP_Param/AP_Param.h:25:0,
                 from ../../libraries/AC_AttitudeControl/AC_AttitudeControl.h:7,
                 from ../../libraries/AC_AttitudeControl/AC_AttitudeControl_Sub.h:6,
                 from ../../libraries/AC_AttitudeControl/AC_AttitudeControl_Sub.cpp:1:
../../libraries/AP_Common/missing/cmath:3:22: fatal error: cmath: No such file or directory
 #include_next <cmath>
                      ^
compilation terminated.

In file included from ../../libraries/AP_Param/AP_Param.h:25:0,
                 from ../../libraries/AC_AttitudeControl/AC_PosControl.h:4,
                 from ../../libraries/AC_AttitudeControl/AC_PosControl_Sub.h:3,
                 from ../../libraries/AC_AttitudeControl/AC_PosControl_Sub.cpp:1:
../../libraries/AP_Common/missing/cmath:3:22: fatal error: cmath: No such file or directory
 #include_next <cmath>
                      ^
compilation terminated.

In file included from ../../libraries/AP_Param/AP_Param.h:25:0,
                 from ../../libraries/AC_AttitudeControl/AC_AttitudeControl.h:7,
                 from ../../libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.h:6,
                 from ../../libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.cpp:1:
../../libraries/AP_Common/missing/cmath:3:22: fatal error: cmath: No such file or directory
 #include_next <cmath>
                      ^
compilation terminated.
Waf: Leaving directory `/home/osboxes/ardupilot/build/Pixhawk1'
Build failed
 -> task in 'objs/AC_AttitudeControl' failed (exit status 1):                                                                                                
        {task 139683473623712: cxx ControlMonitor.cpp -> ControlMonitor.cpp.0.o}                                                                             
 (run with -v to display more information)                                                                                                                   
 -> task in 'objs/AC_AttitudeControl' failed (exit status 1):                                                                                                
        {task 139683473623472: cxx AC_AttitudeControl_Sub.cpp -> AC_AttitudeControl_Sub.cpp.0.o}                                                             
 (run with -v to display more information)                                                                                                                   
 -> task in 'objs/AC_AttitudeControl' failed (exit status 1):                                                                                                
        {task 139683473623592: cxx AC_PosControl_Sub.cpp -> AC_PosControl_Sub.cpp.0.o}                                                                       
 (run with -v to display more information)                                                                                                                   
 -> task in 'objs/AC_AttitudeControl' failed (exit status 1):                                                                                                
        {task 139683473623352: cxx AC_AttitudeControl_Heli.cpp -> AC_AttitudeControl_Heli.cpp.0.o}                                                           
 (run with -v to display more information) 
>

I tried removing and re-installing gcc-arm-none-eaib, but with no use, I tried to use the compiler in the website, still no use, I installed and removed cmake, g++, and still all efforts are in vain... And nothing changed. I am pretty sure the issue is something with the compiler, but I don't know where exactly.

I tried another path, which is to configure the waf to change the borad to pxf, and this is the result I got:
> Setting top to                           : /home/osboxes/ardupilot 
Setting out to                           : /home/osboxes/ardupilot/build 
Autoconfiguration                        : enabled 
GCOV code coverage analysis              : no 
Setting board to                         : pxf 
Using toolchain                          : arm-linux-gnueabihf 
Checking for 'g++' (C++ compiler)        : not found 
Checking for 'clang++' (C++ compiler)    : not found 
could not configure a C++ compiler!
(complete log in /home/osboxes/ardupilot/build/config.log)                                                                                                   
>
Which means it cannot even find a C++ compiler, which doesn't make senes because I have already installed them. (I also did the step of adding the bin file of gcc-none-arm-eabi to my directory, but it's all the same).

I would be grateful if you could help me!
Thank you!

#### Platform
[  ] All

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

proficnc, 2019-05-11:
Please ask for help in the forums 

If you are building for PXF, then it is a Linux target.

PXF is a PXF, nothing to do with FMUv....

closed 2019-05-11 by proficnc

MastaMinds, 2019-05-11:
I tried but didn't know where to post it...
I am mainly building for Pixhawk, (I think mine should be Pixhawk1)
However, the code builds normally for SITL.

auturgy, 2019-05-11:
First thing I’d suggest is run the install script again. Tools/environment_install/install-prereqs-ubuntu.sh -y
Second suggestion is to move this discussion to the forum, which is much better for user support. discuss.ardupilot.org

MastaMinds, 2019-05-11:
Already in the forums.
Not a single reply yet...
I ran the script several times, with no success...

robemorin, 2019-05-14:
I'm stuck too, I solve partially the problem. Change <cmath> for <math.h> in ../AC_PID/AC_PID_2D.h and AP_Math/AP_Math.h

My new problem is <type_traits> library

nikotzoumas, 2022-01-27:
Any updates on this please? Would be a great help.

MastaMinds, 2022-01-27:
> Any updates on this please? Would be a great help.

Hi, I solved this issue by using a different version of Ubuntu (I think it was 16). I really don't remember exactly what happened.
Try using an older version of Ubuntu, and install all the required compilers

manicken, 2022-12-20:
sudo apt-get install g++-arm-linux-gnueabihf
fixed it for me