ArduPilot/ardupilot #11328: Feature request - UPIXELS UP-Flow sensor

FLYBYME, 2019-05-11:
### Feature request

I would like to officiously request support for the "UPIXELS UP-Flow sensor"
https://www.aliexpress.com/item/UPIXELS-UP-FLOW-Optical-Flow-Module-FPV-Drone-Hovering-Altitude-Hold-Control-Optical-Flow-Sensor-For/32969297064.html

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

auturgy, 2019-05-12:
If someone picks this up: data format is posted in the forum https://discuss.ardupilot.org/t/cheerson-cx-of-low-cost-optical-flow-sensor-testing/35598/90

bys1123, 2019-11-29:
[image]
[image]

rmackay9, 2021-03-23:
support added to master with PR https://github.com/ArduPilot/ardupilot/pull/16883 so closing.  txs!

closed 2021-03-23 by rmackay9