ArduPilot/ardupilot #11331: Notch filter is only on first gyro

andyp1per, 2019-05-12:
### Bug report

The notch filter is only run on the first gyro. T

**Issue details**

his leads to a few problems:
- "Gyros inconsistent" errors if you have INS_USE2=1 or INS_USE3=1 since the consistency checks are run on the filtered gyro values.
- If you actually want to use more than one gyro the filtering will be different which seems bad for EKF switching
- There is no notch filter on the accelerometers. Maybe not a bad thing, but the documentation should be fixed.

I am happy to work on this issue if someone can tell me what the right fix is. Options are:
- Have a notch filter per gyro
- Run consistency checks on pre-filtered values
- Turn off consistency checks when using the notch filter.

I think the first option is the right one, but happy to go with other opinions.

**Version**
3.6.9rc2

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Pixracer

**Logs**

magicrub, 2019-05-12:
@priseborough ping

priseborough, 2019-05-15:
EKF should not be using notch filtered gyro rates - is uses delta angles which is a different data product. The notch filtered data product is intended for use by the control loops - @lthall to confirm.

lthall, 2019-05-16:
Yes, the notch filter is only used in the control loops. But there are other issues that need to be addressed as described in: https://discuss.ardupilot.org/t/notch-filter-is-not-a-notch-filter/42280

closed 2019-05-16 by lthall