ArduPilot/ardupilot #11341: Altitude hold controller feed forward acceleration non-zero when controller is being relaxed

tscao, 2019-05-13:
When we relax the altitude hold controllers, we set the target altitude to our current altitude, as well as the target vertical velocity (and also the last target velocity used for computing the feedforward acceleration) to the current vertical velocity, see [this piece of code](https://github.com/ArduPilot/ardupilot/blob/7b7bdd9bf8f25ee63344276890a7bfb2c3008b05/libraries/AC_AttitudeControl/AC_PosControl.cpp#L372).
However, when calling `run_z_controller` afterwards, the target velocity is overwritten with `0`, because our target altitude equals our current altitude. This results in a non-zero feed-forward acceleration, which might not be what we want when the vehicle waits on the ground.
I am not sure however if this was made on purpose, for instance if the vehicle waits on a moving platform (we would need to set our target velocity to the velocity of the platform=the velocity of the vehicle as long as it is waiting on the platform).

IamPete1, 2021-08-29:
@lthall I think this has been resolved, or is by design?

lthall, 2021-08-29:
Yep

closed 2021-08-29 by lthall