ArduPilot/ardupilot #11343: Copter 3.6 and up : able to import 3D object for avoidance

roque-canales, 2019-05-13:
Hello,

Idea is to import 3D object for avoidance without sensor.

Just with GPS, for example , I enter the 3D volume of one tree, and copter can't go in.

Is this feature exist ?
If this feature not exist maybe with companion computer it's possible to send 3D objects for avoidance ?

rmackay9, 2019-05-14:
We don't support this but I think it is very similar to stay-out zones.  The possible addition here is the vertical axis.. stay out zones are generally 2D with infinite height.

IamPete1, 2021-08-15:
@rishabsingh3003 I think we can do this now?

rishabsingh3003, 2021-08-15:
Yes, it is! Via OBSTACLE_DISTANCE_3D mavlink message. Fences are still 2D though .. or rather Horizontal + Vertical

rmackay9, 2021-08-16:
TBH, I don't think that the obstacle-distance-3d is really for the purpose specified by this issue.

peterbarker, 2021-08-17:
On Sun, 15 Aug 2021, Randy Mackay wrote:

> TBH, I don't think that the obstacle-distance-3d is really for the purpose specified by this issue.

Agreed.  It's more like the opposite of the mavlink SAFETY_ALLOWED_AREA 
concept.  Which is a volume, not an area....

peterbarker, 2021-08-17:
.... but @rmackay9 's suggestion of alt limits for separate polygon fences is probably a better way forward.  if rather computationally expensive.

manavgandhi17, 2022-03-08:
Pinging back on this to see if there has been any developmental work done on fence-based 3D obstacle avoidance. Would be very helpful for sectioning out trees instead of an infinite vertical cylinder.

We would probably also need to integrate altitude type into this distinction (MSL, AGL, Terrain) - as well has somehow configuring the same altitude value/type for all vertices of a singular polygon.