ArduPilot/ardupilot #11350: copter 3.6.8 Zed-f9p gps time error

roque-canales, 2019-05-14:
hello,

Trying to make z9p work in UBX, position and vel is ok, but GPS Time Fail….

This is due to that is not possible to activate SOL msg, so I tryed with GPS TIME and GPS UTC, but no work.

Maybe you can update this on next minor release?

Also when using it in NMEA there is also errors on GPS time….and EK2 need 3D vel to good work. So NMEA is not useable.

hdtechk, 2019-05-14:
There is an open PR to fix this already #10951

roque-canales, 2019-05-25:
Randy does this enhancement is already fixed?
If not, can it be added to minor release ?

peterbarker, 2020-11-25:
These units should work fine on our current stable and master branches.  Please open a new issue if they are not.

closed 2020-11-25 by peterbarker