ArduPilot/ardupilot #11351: Copter: crop sprayer liquid level monitor and failsafe

rmackay9, 2019-05-15:
We should add a crop spraying liquid level monitor.  This could either be similar to an ICE engine fuel level monitor (some users use the RSSI driver's pwm input for this) or perhaps something new.  It might even be implemented as a type of battery monitor.

Once the liquid level gets low it could:

- record the vehicle's current position in eeprom (so it is available even if the vehicle is powered down).  This position might just be a lat/lon/alt but if the vehicle is in Auto mode it might need to include the active command's index in the mission.
- trigger a failsafe (i.e. RTL to home)

This request is quite similar to another request to allow users to resume a mission in case their vehicle's battery failsafe has triggered mid-mission.  I.e. the battery failsafe triggers, the vehicle RTL's home, the user replaces the battery and then resumes the mission.  In the past, some have suggested that this "resume" portion is best handled by the ground station instead of in the flight code.

AeroScanTech, 2019-05-15:
There is also the option of maintain the Pixhawk energized, while changing the batteries and adding product (liquid).
It could be an alternate battery that the user plug in before disconnecting the main power just 5v to keep the pixhawk alive, then when main power is replaced the aux battery is removes and then the user can resume the mission.

Rusty105, 2019-05-17:
Would this be better served on a companion computer? Leave flight control to the flying duties and have the companion computer play the role of "mission specialist" and watch over the spraying system?

melgueta, 2020-02-12:
@rmackay9 I would like to contribute on this project! I'm new in this software autopilot development, but skilled programming. Few months ago I integrated the Pixhawk Autopilot to control de GAIA 160 AG (22.5 Lts tank). Despite it works very well, this feature development in pixhawk is very necesary. Specially the Failsafe for liquid refill and continue the mission from last position.

jlcortex, 2020-04-14:
Interesting feature!

Sankalpa12345, 2020-06-12:
@melgueta 
Im too running a similar setup with 24.5 L rank. I Would Love to know what you did and how you did. Im struggleing with this for almost a ear now... its a great pleasure if you could help out to do the Mission Continuation after Battry Swap and Liquid Refill ....
Thank You