ArduPilot/ardupilot #11359: Floating exception error caused by changing certain parameters' value

ktskt, 2019-05-16:
### Bug report

**Issue details**
Hello, developers.

I found multiple floating point exception issue by changing parameters
I put each parameter and value. And, I describe its error message as follows.
I used some extreme values. 
But, if this exception happens, it indicates that there is a potential problem in the certain condition.
So, I think it should be patched.

Also, I think it is necessary to put the range check against unreasonable values IMO.

1. param set PSC_VELXY_I -10000000000000000000000000
Program received signal SIGFPE, Arithmetic exception.
0x000000000045abed in Vector2<float>::length() const ()
(gdb) info stack
#0  0x000000000045abed in Vector2<float>::length() const ()
#1  0x00000000004b9455 in AC_PID_2D::get_i_shrink() ()
#2  0x00000000004b5361 in AC_PosControl::run_xy_controller(float, float) ()
#3  0x00000000004b5f44 in AC_PosControl::update_xy_controller(float) ()
#4  0x00000000004be38e in AC_WPNav::update_wpnav() ()
#5  0x000000000041ef52 in Copter::ModeGuided::takeoff_run() ()
#6  0x00000000004072ae in Copter::fast_loop() ()
#7  0x000000000046defe in AP_Scheduler::loop() ()
#8  0x0000000000406a30 in Copter::loop() ()
#9  0x00000000004d0e30 in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
#10 0x000000000040339e in main ()

2. param set ATC_RAT_RLL_D -10000000000000000000000000
Program received signal SIGFPE, Arithmetic exception.
0x00000000004af251 in AC_AttitudeControl::control_monitor_update() ()
(gdb) info stack
#0  0x00000000004af251 in AC_AttitudeControl::control_monitor_update() ()
#1  0x000000000040727d in Copter::fast_loop() ()
#2  0x000000000046defe in AP_Scheduler::loop() ()
#3  0x0000000000406a30 in Copter::loop() ()
#4  0x00000000004d0e30 in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
#5  0x000000000040339e in main ()

3. param set ATC_RAT_YAW_D -10000000000000000000000000
Program received signal SIGFPE, Arithmetic exception.
0x00000000004af2f2 in AC_AttitudeControl::control_monitor_update() ()
(gdb) info stack
#0  0x00000000004af2f2 in AC_AttitudeControl::control_monitor_update() ()
#1  0x000000000040727d in Copter::fast_loop() ()
#2  0x000000000046defe in AP_Scheduler::loop() ()
#3  0x0000000000406a30 in Copter::loop() ()
#4  0x00000000004d0e30 in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
#5  0x000000000040339e in main ()

4. param set ATC_RAT_PIT_D -10000000000000000000000000
Program received signal SIGFPE, Arithmetic exception.
0x00000000004af2b7 in AC_AttitudeControl::control_monitor_update() ()
(gdb) info stack
#0  0x00000000004af2b7 in AC_AttitudeControl::control_monitor_update() ()
#1  0x000000000040727d in Copter::fast_loop() ()
#2  0x000000000046defe in AP_Scheduler::loop() ()
#3  0x0000000000406a30 in Copter::loop() ()
#4  0x00000000004d0e30 in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
#5  0x000000000040339e in main ()

5. param set PSC_VELXY_P    -10000000000000000000000000
Program received signal SIGFPE, Arithmetic exception.
0x00000000004b54df in AC_PosControl::run_xy_controller(float, float) ()
(gdb) info stack
#0  0x00000000004b54df in AC_PosControl::run_xy_controller(float, float) ()
#1  0x00000000004b5f44 in AC_PosControl::update_xy_controller(float) ()
#2  0x00000000004be38e in AC_WPNav::update_wpnav() ()
#3  0x000000000041ef52 in Copter::ModeGuided::takeoff_run() ()
#4  0x00000000004072ae in Copter::fast_loop() ()
#5  0x000000000046defe in AP_Scheduler::loop() ()
#6  0x0000000000406a30 in Copter::loop() ()
#7  0x00000000004d0e30 in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
#8  0x000000000040339e in main ()

6. param set ATC_RAT_RLL_P  -10000000000000000000000000
Program received signal SIGFPE, Arithmetic exception.
0x00000000004af229 in AC_AttitudeControl::control_monitor_update() ()
(gdb) info stack
#0  0x00000000004af229 in AC_AttitudeControl::control_monitor_update() ()
#1  0x000000000040727d in Copter::fast_loop() ()
#2  0x000000000046defe in AP_Scheduler::loop() ()
#3  0x0000000000406a30 in Copter::loop() ()
#4  0x00000000004d0e30 in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
#5  0x000000000040339e in main ()

7. param set ATC_RAT_PIT_P  -10000000000000000000000000
Program received signal SIGFPE, Arithmetic exception.
0x00000000004af28e in AC_AttitudeControl::control_monitor_update() ()
(gdb) info stack
#0  0x00000000004af28e in AC_AttitudeControl::control_monitor_update() ()
#1  0x000000000040727d in Copter::fast_loop() ()
#2  0x000000000046defe in AP_Scheduler::loop() ()
#3  0x0000000000406a30 in Copter::loop() ()
#4  0x00000000004d0e30 in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
#5  0x000000000040339e in main ()

8. param set ATC_ANG_RLL_P  -10000000000000000000000000
ERROR: Floating point exception - aborting (I changed parameter during AUTO mode)

9. param set ATC_ANG_PIT_P  -10000000000000000000000000
ERROR: Floating point exception - aborting (I changed parameter during AUTO mode)

10. param set ATC_ANG_YAW_P  -10000000000000000000000000
ERROR: Floating point exception - aborting (I changed parameter during AUTO mode)

11. param set ATC_RAT_RLL_FF -10000000000000000000000000
ERROR: Floating point exception - aborting (I changed parameter during AUTO mode)

12. param set ATC_RAT_YAW_FF -10000000000000000000000000
ERROR: Floating point exception - aborting (I changed parameter during AUTO mode)

**Version**
ArduCopter 3.6.9rc2 (the most recent stable 3.6 version)

**Platform**
[  ] All
[  ] AntennaTracker
[O ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
SITL

Jaaaky, 2019-05-17:
Yes, I can confirm this on plane too. Caught one yesterday.

rmackay9, 2019-06-25:
We discussed this on the dev call and a few points came out of this:

- this seems like the output from a [Fuzz Test](https://en.wikipedia.org/wiki/Fuzzing).  These tests can be a useful way to find issues where bad user input could crash the system
- we should be careful to distinguish between a **vehicle crash** and an **unrecoverable flight controller lock-up**.
  - many parameter changes could cause a vehicle to crash.  For example changing the AHRS_ORIENT to 8 (roll 180) will cause the EKF to think the vehicle is upside down and it will certainly crash.  Similarly setting an attitude control gain to a very high number will often cause the vehicle to oscillate so badly that it crashes.  Because AP supports such a wide variety of vehicles it is likely impossible to protect against all these possible failures.
  - we should never have an unrecoverable flight controller lock-up.  This is a situation where the flight controller stops responding to input completely.  A floating point exception could cause this but we handle these exceptions different in SITL vs a real-vehicle.  To simulate what happens on a real-vehicle (in SITL) set the SIM_FLOAT_EXCEPT parameter to 0.

So @ktskt could you try testing again after setting SIM_FLOAT_EXCEPT = 0?

In general we are keen to find unrecoverable flight lock-ups but less interested in param changes that cause the vehicle to crash.

IamPete1, 2021-09-12:
I'm going to close this, please re-open if your able to re-test with `SIM_FLOAT_EXCEPT` = 0. We would also be interested if you can give us some information on how you ran the testing, it might be something we cold include as part of AP's autotesting.

closed 2021-09-12 by IamPete1

figuernd, 2024-06-18:
Param testing seems to be an issue with pursuing. Just ran into an ardusub bug failing with a floating point exception.

After doing some gdb tracing found that `AP_Logger_File::periodic_1Hz()` relies on `FILE_RATEMAX` and `DISARM_RATEMAX`, which do not have defaults. Setting those to 0 fixed.