ArduPilot/ardupilot #11369: Copter: Switching off from AUTO-mode while climbing causes high desired climb rate (3.6.8 ChibiOS)

VDLJu, 2019-05-17:
#### Issue details
Flight mode changed off from AUTO causes a abnormal desired climb rate while the drone is climbing. Level flight and descending doesn't trigger this condition.

[image]

Also the autopilot is not able to follow the desired altitude in AUTO modes, but works nicely in Loiter and AltHold modes.

[image]

EKF and sensors are working properly. This behavior is highly likely a some sort of bug or a edge case that should be handled on code.

Same hardware with same parameters running on copter 3.6.0 (NuttX) works properly.

#### Version
This issue was encountered after update from copter 3.6.0 (NuttX) to copter 3.6.8 (ChibiOS).

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Used frame is X4 copter. ~5kg. 

#### Hardware type
2018 cube black (PH2.1), HERE GNSS

#### Logs
[Log.zip](https://github.com/ArduPilot/ardupilot/files/3191494/Log.zip)

rmackay9, 2019-05-17:
I think we've had this reported before:

- [Copter: throttle spike if user switches from Auto takeoff to Loiter](https://github.com/ArduPilot/ardupilot/issues/10072)
- [Climb rate and Thr.out issue when switching from RLT to Loiter](https://github.com/ArduPilot/ardupilot/issues/7435)

I think we may have a small improvement in master that will go out with Copter-3.7 but I think I also remember from a previous discussion that the user was doing something unusual that made the altitude control very bad.  I bring this up because one thing that is odd about the 2nd graph linked above is that the desired Alt and actual altitude are not sticking together.

AlvaroRAS, 2019-05-23:
Something like this hapened to me also; the copter was flying in auto mode a mission that consist in ascending to 70m followed by CONDITIONAL_DELAY function (which dosent work) and finally a descending command;  but when the pilot cancel the mission invoking Loiter mode when descending, the controller demand a high desired climb rate spike.

[image]

[image]

These cause a full throttle demand on all motors, causing de-synchronization of motors; the copter can’t maintain its attitude on these situation and crash (totally lost of components).

[image]

[image]

**Version**
Copter 3.6.7 (ChibiOS).

**Platform**
[ ] All
[ ] AntennaTracker
[X] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

**Airframe type**
Hexacopter.

**Hardware type**
Controller:                      2018 Pixhawk CUBE Black and dual HERE GPS.
Weigh:                           13Kg.
Power source:                12S 16Ah Li-Po.
Motor, Propellers, ESC:  KDE Direct Combo.

**Log**

[CASE 2.zip](https://github.com/ArduPilot/ardupilot/files/3214428/CASE.2.zip)

rmackay9, 2019-05-23:
@AlvaroRAS, re the CONDITIONAL_DELAY command, that delay will only apply to "DO" commands that follow it.  It's not a navigation command so it doesn't apply to the following waypoint.  To delay at a waypoint then it's best to fill in the delay field with a number of seconds or the NAV_DELAY command can be used.

AlvaroRAS, 2019-05-23:
Yeah, we notice that, so we cancel the mission at that moment, now we do the panoramic pictures making another waypoint with delay in the same place so the copter stays in hover until the panoramic photo is completed.

[image]

Pedals2Paddles, 2019-05-24:
Anyway, the log does seem to indicate a small increase in desired altitude, and a corresponding monstrous increase in throttle output.  Throttle stick was centered.  Certainly appears to be the same general problem as noted in the other linked issues.

[image]

Kind of related note... if going to full throttle causes all the ESCs to desync, I think I'd be looking for some different ESCs that work better. Not that it changes this problem, but I wouldn't use those ESCs anymore for sure.

VDLJu, 2019-05-24:
I can quite easily and safely replicate the flights if needed.

It's good to hear that the throttle spike is worked on. Let me know if there is something I can do to help on the issue with "Alt and actual altitude are not sticking together".

lthall, 2021-09-27:
All this code has been rewritten in 4.1 with more stringent initialisation.

If this is still a problem we can reopen.

closed 2021-09-27 by lthall